openrazer/openrazer #1939: No devices detected for Huntsman V2 Analog

revenz, 2022-10-27:
Hi, I'm getting no devices detected when I have a huntsman v2 analog plugged in.

some output
```
[john@batman ~]$ openrazer-daemon --version
openrazer-daemon 3.4.0
[john@batman ~]$ lsusb | grep 1532
Bus 013 Device 004: ID 1532:0266 Razer USA, Ltd Razer Huntsman V2 Analog
```

z3ntu, 2022-10-27:
Can you send the output of `sudo modprobe razerkbd` and `lsmod | grep razer` ?

revenz, 2022-10-27:
```
[john@batman ~]$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Key was rejected by service
[john@batman ~]$ lsmod | grep razer
[john@batman ~]$ 

```

z3ntu, 2022-10-27:
I'm not 100% sure but sounds like Secure Boot is making this not work.

See https://github.com/openrazer/openrazer/wiki/Troubleshooting#3-kernel-modules-do-they-load-properly

> The easiest solution is disabling Secure Boot in the BIOS/UEFI. The complicated solution is signing the built kernel modules, so that the kernel accepts them. See [#98](https://github.com/openrazer/openrazer/issues/98) or just use your favorite search engine.

Definitely worth a shot.

revenz, 2022-10-27:
thanks, that was it!

closed 2022-10-27 by revenz