openrazer/openrazer #1942: Signed software

TECH7Fox, 2022-11-04:
Would be great if people didn't need to disable Secure boot or sign the modules themselve. Is there any reason that this isn't the case?

z3ntu, 2022-11-04:
You can sign the modules yourself and setup secure boot with it, it's just that no distro I know of has made this somewhat sane to do. So I also can't really support it given if you do it, it's some kind of custom setup.

See also https://github.com/openrazer/openrazer/wiki/Troubleshooting#3-kernel-modules-do-they-load-properly

> The complicated solution is signing the built kernel modules, so that the kernel accepts them. See [#98](https://github.com/openrazer/openrazer/issues/98) or just use your favorite search engine.

TECH7Fox, 2022-11-04:
Yeah I saw that. But for now I just disabled secure boot, although my system really didn't like that.

But it would be nice if it could be "officialy" signed. Would make the install much easier. Is it possible to sign them here in the repo?

lah7, 2022-11-04:
Since OpenRazer is an out-of-tree driver in the form of source code, it gets built into the kernel during installation (and repeats again when the kernel is upgraded). For most distros, that's done by DKMS. Right now, there's nothing signing that binary when its built so the system doesn't load this "untrusted" module while secure boot is on.

It seems difficult for OpenRazer to sign anything, since we're not distributing a kernel or binary. Everyone has different distros, kernels and versions. It's also less convenient because every user would have to make sure their motherboard trusts this custom key used to sign.

I suppose a distro maintainer _might_ have more leverage, since they _could_ create a compiled package with the binary signed and already trusted alongside their official kernels, but `openrazer` isn't in many official repos to begin with.

It does seem complicated, though Debian has some good documentation on [Secure Boot](https://wiki.debian.org/SecureBoot).

TECH7Fox, 2022-11-04:
Ah, didn't think about that.

Thanks for the explanation and link, makes sense now. Hope that it get's added in offical repos someday then. It is great software. Would also mean that GUI's like polychromatic could more easily be added to stores right?

Anyways, thanks again. 👍

closed 2022-11-04 by TECH7Fox

lah7, 2022-11-07:
No problem. I updated the [Troubleshooting wiki page](https://github.com/openrazer/openrazer/wiki/Troubleshooting#3-kernel-modules-do-they-load-properly) to help clarify secure boot.

> Would also mean that GUI's like polychromatic could more easily be added to stores right?

I guess so.