openrazer/openrazer #1945: Support for Razer Blade 15 Base Model 2022

jleguina, 2022-11-07:
### Device Info

* Name: Razer Blade 15 Base Model 2022
* Product Number: RZ09-04 10AW22
* Model Number: RZ09-0410

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```shell
Bus 003 Device 002: ID 1532:027a Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x027a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

```shell
total 0
drwxr-xr-x 2 root root 160 Nov  7 15:18 .
drwxr-xr-x 4 root root 860 Nov  7 15:18 ..
lrwxrwxrwx 1 root root   9 Nov  7 15:18 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event8
lrwxrwxrwx 1 root root  10 Nov  7 15:18 usb-Razer_Razer_Blade-event-if01 -> ../event17
lrwxrwxrwx 1 root root   9 Nov  7 15:18 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 Nov  7 15:18 usb-Razer_Razer_Blade-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Nov  7 15:18 usb-Razer_Razer_Blade-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 Nov  7 15:18 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

```
Please post the output of
    usbhid-dump -m 1532 -ed
```
```shell
003:002:002:DESCRIPTOR         1667837019.724570
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

003:002:001:DESCRIPTOR         1667837019.725060
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

003:002:000:DESCRIPTOR         1667837019.725410
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

### Packet Captures
Unsure what to do here
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
Coming...
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2022-11-09:
Duplicate of #1581

(CC @meeuw for hid descriptor, btw are you planning to regularly scavenge issues to gather the hid descriptors or how should we handle this?)

meeuw, 2022-11-12:
Oh awesome, someone already shared the hid descriptor!

I've only seen the HID descriptors for two devices so far (this is the third). I don't know how different they are across different devices.

It would be great if we could parse the descriptor in the kernel module and make the driver more generic across devices.

The current in-kernel hid-razer module doesn't parse the hid descriptor either (it uses the same USB mouse assumption as openrazer). I figure it would be best to focus first on the in-kernel razer-hid module (and maybe implement the "Driver model for leds and led triggers" (leds.h)).