openrazer/openrazer #1946: I am using Naga wired X it is not detected on void linux

DirtyVoid, 2022-11-11:
I am using Naga wired X it is not detected on void linux

z3ntu, 2022-11-11:
Did you check our Troubleshooting page? https://github.com/openrazer/openrazer/wiki/Troubleshooting

DirtyVoid, 2022-11-11:
yes, I did and I followed the instructions. I also tried some of the other closed issues nothing worked as of yet as I cannot narrow it down. 
```
> lsusb | grep 1532     
Bus 001 Device 002: ID 1532:0096 Razer USA, Ltd Razer Naga X
```
```
> sudo dkms install openrazer-driver/3.4.0 
Module openrazer-driver/3.4.0 already installed on kernel 5.19.17_1 (x86_64).
```
```
sudo dkms status                        
nvidia/515.65.01, 5.19.10_1, x86_64: installed
nvidia/515.65.01, 5.19.13_1, x86_64: installed
nvidia/515.65.01, 5.19.16_1, x86_64: installed
nvidia/515.65.01, 5.19.17_1, x86_64: installed
openrazer-driver/3.4.0, 5.19.10_1, x86_64: installed
openrazer-driver/3.4.0, 5.19.13_1, x86_64: installed
openrazer-driver/3.4.0, 5.19.16_1, x86_64: installed
openrazer-driver/3.4.0, 5.19.17_1, x86_64: installed
```

```
>  sudo modprobe razerkbd                

```
I was already in plugdev group, but i still ran it
```
> sudo gpasswd -a dirtyv plugdev
Adding user dirtyv to group plugdev
```

```
> lsmod | grep razer
razerkbd               98304  0
usbcore               290816  13 xhci_hcd,snd_usb_audio,usbhid,snd_usbmidi_lib,razerkbd,rtsx_usb,usb_storage,usbkbd,uvcvideo,btusb,usbmouse,xhci_pci,uas
hid                   143360  6 i2c_hid,usbhid,hid_multitouch,razerkbd,hid_generic,uhid

```
it has no output on dmesg
```
> sudo dmesg | grep razer   
```

```
x sudo journalctl -b | grep -i razer_mount 
sudo: journalctl: command not found
```

```
> cat ~/.local/share/openrazer/logs/razer.log 
2022-11-11 02:11:47 | razer                          | INFO     | Initialising Daemon (v3.4.0). Pid: 2237
2022-11-11 02:11:47 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-11-11 02:11:47 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
[... 339 lines cut ...]
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViper8KHz -------------------------- (1532:0091)
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViperMini -------------------------- (1532:008A)
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViperUltimateWired ----------------- (1532:007A)
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViperUltimateWireless -------------- (1532:007B)
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViperV2ProWired -------------------- (1532:00A5)
2022-11-11 02:13:35 | razer                          | DEBUG    | Loaded device specification: RazerViperV2ProWireless ----------------- (1532:00A6)
2022-11-11 02:13:35 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
[... cut ...]
2022-11-11 02:13:35 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2022-11-11 02:13:35 | razer                          | INFO     | Serving DBus
2022-11-11 02:14:28 | razer                          | DEBUG    | DBus called get_serial_list
2022-11-11 02:19:47 | razer                          | INFO     | Stopping daemon on signal 1
2022-11-11 02:19:47 | razer                          | DEBUG    | Writing persistence config

```

```
> ls -al /sys/bus/hid/drivers/razer*/*1532*/
zsh: no matches found: /sys/bus/hid/drivers/razer*/*1532*/
```

z3ntu, 2022-11-11:
For you this part of the readme applies:

> The list below contains latest devices supported on this branch (usually master). These might not be released yet, so check the [stable branch](https://github.com/openrazer/openrazer/tree/stable) for what should be available in the packages for your distribution.

Either wait for the next release, or you need to build from git sources.

z3ntu, 2022-11-11:
Or actually also the very first section of the Troubleshooting page: https://github.com/openrazer/openrazer/wiki/Troubleshooting#1-check-device-support

closed 2022-11-11 by DirtyVoid