openrazer/openrazer #1947: Razer Ornata V3 X

Mdievendorf, 2022-11-12:
Razer Ornata V3 X - p/n = RZ03-04470200-R3U1

susb -d 1532: -v                                                  127 ⨯

Bus 002 Device 007: ID 1532:0294 Razer USA, Ltd Razer Ornata V3 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0294 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3 X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

└─$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 240 Nov 11 22:28 .
drwxr-xr-x 4 root root 680 Nov 11 22:28 ..
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Dell_Dell_AC511_USB_SoundBar-event-if03 -> ../event26                                                         
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Nordic_2.4G_Wireless_Receiver-event-if01 -> ../event20                                                        
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Nordic_2.4G_Wireless_Receiver-event-kbd -> ../event17                                                         
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Nordic_2.4G_Wireless_Receiver-if01-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 Nov 11 22:28 usb-Nordic_2.4G_Wireless_Receiver-if01-mouse -> ../mouse0                                                         
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Razer_Razer_Ornata_V3_X-event-if01 -> ../event24                                                              
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Razer_Razer_Ornata_V3_X-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Razer_Razer_Ornata_V3_X-if01-event-kbd -> ../event23                                                          
lrwxrwxrwx 1 root root  10 Nov 11 22:28 usb-Razer_Razer_Ornata_V3_X-if02-event-mouse -> ../event25                                                        
lrwxrwxrwx 1 root root   9 Nov 11 22:28 usb-Razer_Razer_Ornata_V3_X-if02-mouse -> ../mouse1

momjyandavit, 2022-11-27:
I really need support for my keyboard, I am not going back to windows. I just have to turn the backlight off entirely for now.

raiderrising, 2022-12-30:
I also have Razer Ornata V3 X, and would like support. Please! The item is detected but it comes up as a number.
Looking forward to hearing from you.  
[Razer Ornata V3 X _01.odt](https://github.com/openrazer/openrazer/files/10322906/Razer.Ornata.V3.X._01.odt)
[Razer Ornata V3 X _02.o
[image]
dt](https://github.com/openrazer/openrazer/files/10322908/Razer.Ornata.V3.X._02.odt)

rlanos, 2023-01-06:
### Device Info

* Name: Ornata v3 X
* Product Number: RZ03-04470500-R3F1
* Model Number: RZ03-0447

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 002 Device 024: ID 1532:0294 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0294 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 112 lines cut ...]
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

002:023:000:DESCRIPTOR         1673013311.569084
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```
### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
I'll add the synapse screenshots later

mordilos, 2023-02-06:
hello, I also need the support for Ornata V3X and have the same issues as above.

jackwilliams79, 2023-02-17:
I also would like to have my Ornata V3 X keyboard supported by OpenRazer. What is needed to get this going?

jackwilliams79, 2023-03-27:
@MoonlightWave-12 Thank you :+1:

@z3ntu How can this be integrated? What's the next step?

z3ntu, 2023-04-22:
Pcaps look good, thanks! There's not much in the pcaps (which is totally normal nowadays) but transaction_id is 0x1f and MATRIX_DIMS is [1, 1]. With this just follow any recent mouse addition commits in the repo and replicate it for this device!

closed 2023-04-27 by z3ntu