openrazer/openrazer #1948: plugdev group doesn't exist

TheTrashCrafter, 2022-11-12:
### Distribution

Fedora Workstation 36

### Support

- [Yes] Have you tried rebooting?
- [Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,
I'm currently trying to install OpenRazer on Fedora, but even the first step doesn't work: If i try to run this command:
`sudo gpasswd -a $USER plugdev`
It tells me, that there is no plugdev group in /etc/group.

Regards,
Kevin

z3ntu, 2022-11-14:
Weird, the install script from the package you installed should've created it for you: https://github.com/openrazer/OBS-packaging/blob/master/openrazer.spec#L129-L133

Can you try running the `sudo groupadd -r plugdev` manually?

TheTrashCrafter, 2022-11-15:
That resulted in crashing, but i reinstalled Fedora and now it works, thanks for your help.

Regards,
Kevin