openrazer/openrazer #1950: Support for Razer Kraken V3 Pro

AZeroEight, 2022-11-15:
### Device Info

* Name: Razer Kraken V3 Pro
* Product Number: RZ04-03460100
* Model Number: RZ04-0346

### Device Information
```
Bus 001 Device 002: ID 1532:052c Razer USA, Ltd Razer Kraken V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x052c 
  bcdDevice            9.00
  iManufacturer           1 NXP SEMICONDUCTORS
  iProduct                2 Razer Kraken V3 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x012e
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass          1 Audio
      bFunctionSubClass       1 Control Device
      bFunctionProtocol       0 
      iFunction               2 Razer Kraken V3 Pro
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0048
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Microphone
        bAssocTerminal          6
        bNrChannels             2
[... 395 lines cut ...]
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:007:000:DESCRIPTOR         1668523084.525828
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:002:003:DESCRIPTOR         1668523084.526088
 05 0C 09 01 A1 01 15 00 25 01 05 0C 09 CD 09 E9
 09 EA 09 B5 09 B6 09 B7 95 06 75 01 81 02 75 02
 95 01 81 03 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

AZeroEight, 2022-11-17:
This one is too hard for me to fix myself. The device receiving the changes from Synapse is not the HID device part but the CDC part. If you look at the capture you see a value 0xff7f7f being applied - that is the color I gave the headset as static color.

I have no experience in reverse engineering USB or writing drivers.

z3ntu, 2022-11-20:
Based on the packet captures it looks like a fairly standard device, transaction id 0x1f, MATRIX_DIMS = `[1, 22]`. Probably copy-pasting KRAKEN_KITTY_EDITION is a good starting point for this device.

TobiTenno, 2023-01-20:
I have one of these, so i'm happy to test if someone has a branch for it

Romz24, 2023-02-12:
Any news on Razer Kraken V3 Pro support?

arch-fan, 2023-04-20:
Just a reminder comment

szymach, 2023-07-19:
@AZeroEight hate to impose on you, but I was wondering if you are still considering support for V3 Pro? I have just purchased them and the fact that they keep turning RGB on by default is driastically reducing the battery's life. I can still return them, so if this is never going to get supported, I may consider returning them.

z3ntu, 2023-08-06:
See https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md for notes about how to add support yourself.

tiritto, 2023-11-30:
Any progress on this front? I've looked into the abovementioned notes about how to add support yourself, but it's not exactly helping when you have no idea about how it all works in the first place.

paladin-devops, 2024-12-30:
I am a new owner of these headphones and I'd like to look into this. I'm not familiar with programming headsets/peripheral drivers, but I'll be following `DEVELOPMENT.md` closely.

paladin-devops, 2025-01-01:
Still working on this. Something of note that I observed is that the Kraken v3 Pro headset uses the protocol which is used for keyboards.

```
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard  <----
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
```

paladin-devops, 2025-01-02:
@AZeroEight I've got a [work-in-progress PR](https://github.com/openrazer/openrazer/pull/2372) up, still working through some issues.

paladin-devops, 2025-01-06:
I think that @AZeroEight's comment [here](https://github.com/openrazer/openrazer/issues/1950#issuecomment-1319057661) is pertinent:

> This one is too hard for me to fix myself. The device receiving the changes from Synapse is not the HID device part but the CDC part. If you look at the capture you see a value 0xff7f7f being applied - that is the color I gave the headset as static color.

As can be seen in [my pull request](https://github.com/openrazer/openrazer/pull/2372), I attempted to add this device to the `razerkraken` driver. (after initially trying to add it to the `razeraccessory` driver (to mimic the [Razer Kraken Kitty Edition implementation](https://github.com/openrazer/openrazer/pull/1262)), which I squashed when I changed my approach) I think that based on what I have learned about the Kraken V3 Pro, it does not conform to the structure of the driver that has been written for the other devices supported there, such as the Kraken 7.1. As mentioned in the `lsusb -v -d 1532:052c` output in OP, there are several interfaces for this device, with the HID interface being one of them (#3). However, _alll_ requests sent by `usb_control_msg` ([here](https://github.com/openrazer/openrazer/blob/5e677b178be7da08fb72d7cd0791bf68a129fcc5/driver/razerkraken_driver.c#L62-L69)) fail with a -EPIPE error. This happens even with what I'd describe as the simplest of operations, [fetching the device serial number](https://github.com/openrazer/openrazer/blob/5e677b178be7da08fb72d7cd0791bf68a129fcc5/driver/razerkraken_driver.c#L557). I suppose it's also worth mentioning that the device is reset to its factory condition, which I did by using Synapse on a Windows machine. I also updated the firmware to the latest, v1.10.0. To experiment, with `usb_control_msg`, I tried updating the driver to send the request to the HID interface of this dvice, interface # 3. I also modified the protocol for the device to be the keyboard protocol, because that is the protocol mentioned for the HID interface in the device descriptor:

<details>
  <summary>HID Interface Info</summary>
  ```shell
Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
  ```
</details>

I ran into similar broken pipe errors when implementing this in the `razeraccessory` driver, [which I re-did in a different branch](https://github.com/paladin-devops/openrazer/commit/92c11b81f23c212dc1518abd9d3b775fb6d6af0b). For this to begin to work, [this line](https://github.com/openrazer/openrazer/blob/5e677b178be7da08fb72d7cd0791bf68a129fcc5/driver/razeraccessory_driver.c#L2074) must be updated to accept interface 3, because interface 0 is not HID. However, even after doing this, the result is still a -EPIPE (for all requests, including fetching the serial number).

I am out of time to tinker for a bit, but I wanted to share this update for anyone who may have more experience or insight on this subject. I managed to reverse engineer one of the BULK OUT requests with this crude utility (below), with the hope that I may begin to add support for such requests to a COM interface of this device, instead of HID, which is the primary protocol that OpenRazer seems to support.

<details>
  <summary>Utility for sending BULK OUT request</summary>
#include <stdio.h>
#include <string.h>
#include <libusb-1.0/libusb.h>
#include <errno.h>

#define VENDOR_ID 0x1532  // Razer's vendor ID
#define PRODUCT_ID 0x052c

int main() {
    libusb_device_handle *dev_handle = NULL;
    libusb_context *ctx = NULL;
    int r;

    // Initialize libusb
    r = libusb_init(&ctx);
    if (r < 0) {
        fprintf(stderr, "Failed to initialize libusb: %s\n", libusb_error_name(r));
        return 1;
    }

    // Open device with Vendor and Product ID
    dev_handle = libusb_open_device_with_vid_pid(ctx, VENDOR_ID, PRODUCT_ID);
[... 62 lines cut ...]
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x6d
    };

    // Send the bulk transfer, assuming endpoint 2 is correct for this device. 
    // You might need to adjust the endpoint if this is not correct.
    int transferred = 0;
    r = libusb_bulk_transfer(dev_handle, (6 | LIBUSB_ENDPOINT_OUT), data, sizeof(data), &transferred, 1000);
    if (r == 0 && transferred == sizeof(data)) {
        printf("Data sent successfully!\n");
    } else {
        fprintf(stderr, "Failed to send data. Error: %s. Transferred %d bytes. errno: %d\n", libusb_error_name(r), transferred, errno);
    }

    libusb_release_interface(dev_handle, interface_number);

    // Close device and exit libusb
    libusb_close(dev_handle);
    libusb_exit(ctx);

    return 0;
}

</details>

Side note, thanks to @z3ntu for maintaining this project!

paladin-devops, 2025-01-19:
Some additional, possibly relevant information about this device is that there is a Wireless USB dongle, which is used to communicate with the headset. That USB information was already provided (for product ID `052c`). When plugging in the headset directly (which charges the device), the product ID is `052d`, and here's the USB information for that:

<details>
  <summary>052d USB info</summary>

```shell
$ lsusb -d 1532:052d -v

Bus 001 Device 013: ID 1532:052d Razer USA, Ltd USB COMPOSITE VCOM
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x052d 
  bcdDevice            9.00
  iManufacturer           1 Razer Barracuda X
  iProduct                2 USB COMPOSITE VCOM
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               3 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x01
          call management
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

```
</details>

It seems that based on the packet capture, the Razer Synapse application is indeed using URB BULK requests to communicate with the wireless USB dongle. Based on [the product information](https://www.amazon.com/Razer-HyperSense-Wireless-Headset-Technology/dp/B09HJB6R4Z), the dongle communicates with the headset wirelessly via the Razer HyperSpeed 2.4 GHz protocol (not Bluetooth). Perhaps that uses HID communication under the hood, but it's impossible to say, as I cannot capture the traffic between the dongle and the headset itself. I will abandon the open PR I have, and try a different approach.