openrazer/openrazer #1954: Kernel dependencies and updates

lelaut1, 2022-11-18:
### Distribution

### Support

- [x ] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello everyone,

I'm new to linux and after trying my best to install open razer for my DeathAdder V2 X HyperSpeed. I'm running into dependencies problems. I've tried to go after the installation guide and enable "universe" in the update configs but that didn't help.  
Right now I have the problem that I can't update any software because of broken packages. 
Here's the error message:

[image]
It says: "please check if you have any package sources from third parties. If that is the case please remove them. Try : apt-get install -f
Transaction failed: package system is damaged. Not fulfilled dependencies."
When I type in the command it ties to install linux-image-unsigned-5.15.0-50-generic and to remove linux-image-5.15.0-50-generic. Here a picture of the process:

[image]
I've also tried the Troubleshooting but I had no problems with it except that my device didn't showed up in the list. 
 
I hope somebody can help me, so that I'm able to update my distribution again.

lah7, 2022-11-19:
Hello. Which version of Ubuntu are you using? It looks your GRUB (bootloader) is misconfigured and prevents your system from being updated. 

>  /usr/sbin/grub-mkconfig: 10: /etc/default/grub: splash: not found

A search for this brings up this, I would check that file:

* https://askubuntu.com/questions/950331/grub-error-usr-sbin-grub-mkconfig-7-etc-default-grub-splash-not-found

After that's resolved, these commands may help:

> Configure packages that were unpacked but not configured:
>
>     sudo dpkg --configure -a
> 
> Download and install broken dependencies:
>
>     sudo apt --fix-broken install
>
> _(Taken from Ubuntu MATE Welcome)_

If this doesn't work, you may have better luck asking at a place like [Ubuntu Forums](https://ubuntuforums.org/) who are probably more familiar with this kind of problem.

lelaut1, 2022-11-19:
I'm using Ubuntu 22.04.1 LTS 64bit.  Genome  42.5.
The Link helped and i was able to update my device again. Thanks a lot for that! 
But for some reason it fails to download software package source information when i go to the update settings. When it comes to the path /razer/ubuntu20.1 it fails and after 2 retries and a long time of waiting at the third try,  it manages to show me all the available update except for razer. 
Is there a way to uninstall the drivers, kernels and packages again?

lah7, 2022-11-20:
> But for some reason it fails to download software package source information when i go to the update settings. 

If you mean **Software & Updates**, go to the "Other Software" tab and uncheck everything there. You can try deleting whatever `/razer/ubuntu20.1` is, if you see it in the list. Not sure what that could be. After clicking "Close", it will prompt to refresh the _software cache / package source info_.

This package source information download error could happen if the mirror you're using is having problems - you can try another via the first tab ("Ubuntu Software"). 🇬🇧  "Main server" should always work, should you need to try another.

Once the system is happy with everything and is up-to-date, you can try installing OpenRazer again using the instructions [from the website.](https://openrazer.github.io/#ubuntu) By default, it should automatically pick the correct "kernel headers" which is only possible manual step needed.

lelaut1, 2022-11-21:
Thank you for your help!
I followed everyone of your steps and all no other problems accrued. 
Openrazer is also properly installed.  
This issue can be closed now.

closed 2022-11-21 by z3ntu