openrazer/openrazer #1956: Huntsman v2 not recognized

Albusoline, 2022-11-19:
### Distribution

Linux Mint 21

### Support

- [x ] Have you tried rebooting? Yes. 
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Keyboard is still unrecognized. Any ideas?
[image]

z3ntu, 2022-11-19:
Huntsman V2 is only supported since v3.3.0, so it won't work with v3.2.0 that you have installed. Check our website for installation instructions.

Albusoline, 2022-11-19:
Hi.
Thanks for the tip, I just updated to 3.4.0... still not working.
[image]

z3ntu, 2022-11-19:
Have you rebooted after updating? Also please provide `lsusb | grep 1532` output

Albusoline, 2022-11-19:
Rebooted LinuxMint.
Opened Polychromatic again. Device is now showing up as "Razer Huntsman V2", however I get a "Backend Error". When "showing details" I get this text:

_Traceback (most recent call last):
  File ’/usr/lib/python3/dist-packages/polychromatic/middleman.py’, line 165, in get_device
    device = module.get_device(uid)
  File ’/usr/lib/python3/dist-packages/polychromatic/backends/openrazer.py’, line 607, in get_device
    ’id’: ids[rate],
KeyError: 250_

Output of `lsusb | grep 1532`:
_Bus 001 Device 007: ID 1532:026c Razer USA, Ltd Razer Huntsman V2_

z3ntu, 2022-11-19:
For that, see https://github.com/polychromatic/polychromatic/issues/430 . At least the OpenRazer side seems to be working correctly now.

closed 2022-11-19 by z3ntu

Albusoline, 2022-11-19:
Thank you z3ntu.
You've been a great help!
It's working now!