openrazer/openrazer #1960: Openrazer stopped loading properly

fungusgod, 2022-11-21:
### Distribution

Linux Mint 20.3 Cinnamon version 5.2.7
Linux kernel 5.4.0-132-generic

### Support

- [ x ] Have you tried rebooting?
- [ x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After a kernel upgrade, my Polychromatic no longer recognizes my devices.  I attempted troubleshooting and even tried rolling back the kernel.  As part of troubleshooting, here is the return I get from sudo dpkg --configure -a:
`Setting up openrazer-driver-dkms (3.5.0~ubuntu20.04.1) ...
Removing old openrazer-driver-3.5.0 DKMS files...

------------------------------
Deleting module version: 3.5.0
completely from the DKMS tree.
------------------------------
Done.
Loading new openrazer-driver-3.5.0 DKMS files...
Building for 5.4.0-132-generic
Building initial module for 5.4.0-132-generic
Error! Bad return status for module build on kernel: 5.4.0-132-generic (x86_64)
**Consult /var/lib/dkms/openrazer-driver/3.5.0/build/make.log for more information.**
dpkg: error processing package openrazer-driver-dkms (--configure):
 installed openrazer-driver-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 3.5.0~ubuntu20.04.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-meta (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on openrazer-driver-dkms (= 3.5.0~ubuntu20.04.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-daemon (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 3.5.0~ubuntu20.04.1); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package python3-openrazer (--configure):
 dependency problems - leaving unconfigured
sh: 0: getcwd() failed: No such file or directory
Errors were encountered while processing:
 openrazer-driver-dkms
 openrazer-meta
 openrazer-daemon
 python3-openrazer
`
I tried to open the make.log but it was empty.
Can someone point me towards what I am missing?  TIA

lah7, 2022-11-21:
It's possible the headers package for the kernel was missing, or there was a version mismatch during the upgrade between the kernel and the headers package.

Looking at Synaptic, for Ubuntu 20.04, make sure `linux-headers-generic` (currently `5.4.0.132.132`) is installed and up-to-date. Then try building the driver again:

    sudo dkms install openrazer-driver/3.5.0

fungusgod, 2022-11-21:
Thank you so much for trying to help me.  I had already input that command into the terminal.  Here is what I get back:
`shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory

Kernel preparation unnecessary for this kernel.  Skipping...
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory

Building module:
cleaning build area...(bad exit status: 2)
KERNELDIR=/lib/modules/5.4.0-132-generic/build make driver...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-132-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/3.5.0/build/make.log for more information.
`
You can see that a reference is again made to consult make.log.  Initially in my troubleshooting it had five lines that I intended to post here.  But after running different steps, the make.log became blank and has remained so.  I fear I am making the situation worse.

fungusgod, 2022-11-21:
Okay, got to thinking and did some more research.  First I ran this:
`fungusgod@Cryptonomicon:/var/lib/dkms/openrazer-driver/3.5.0/build$ ls -l /usr/src/linux-headers-$(uname -r)
total 1612
drwxr-xr-x 3 root root    4096 Nov 16 18:11 arch
lrwxrwxrwx 1 root root      32 Oct 18 00:19 block -> ../linux-headers-5.4.0-132/block
lrwxrwxrwx 1 root root      32 Oct 18 00:19 certs -> ../linux-headers-5.4.0-132/certs
lrwxrwxrwx 1 root root      33 Oct 18 00:19 crypto -> ../linux-headers-5.4.0-132/crypto
lrwxrwxrwx 1 root root      40 Oct 18 00:19 Documentation -> ../linux-headers-5.4.0-132/Documentation
lrwxrwxrwx 1 root root      34 Oct 18 00:19 drivers -> ../linux-headers-5.4.0-132/drivers
lrwxrwxrwx 1 root root      29 Oct 18 00:19 fs -> ../linux-headers-5.4.0-132/fs
drwxr-xr-x 4 root root    4096 Nov 16 18:11 include
lrwxrwxrwx 1 root root      31 Oct 18 00:19 init -> ../linux-headers-5.4.0-132/init
lrwxrwxrwx 1 root root      30 Oct 18 00:19 ipc -> ../linux-headers-5.4.0-132/ipc
lrwxrwxrwx 1 root root      33 Oct 18 00:19 Kbuild -> ../linux-headers-5.4.0-132/Kbuild
lrwxrwxrwx 1 root root      34 Oct 18 00:19 Kconfig -> ../linux-headers-5.4.0-132/Kconfig
drwxr-xr-x 2 root root    4096 Nov 16 18:11 kernel
lrwxrwxrwx 1 root root      30 Oct 18 00:19 lib -> ../linux-headers-5.4.0-132/lib
lrwxrwxrwx 1 root root      35 Oct 18 00:19 Makefile -> ../linux-headers-5.4.0-132/Makefile
lrwxrwxrwx 1 root root      29 Oct 18 00:19 mm -> ../linux-headers-5.4.0-132/mm
-rw-r--r-- 1 root root 1620108 Oct 18 00:19 Module.symvers
lrwxrwxrwx 1 root root      30 Oct 18 00:19 net -> ../linux-headers-5.4.0-132/net
lrwxrwxrwx 1 root root      34 Oct 18 00:19 samples -> ../linux-headers-5.4.0-132/samples
drwxr-xr-x 8 root root   12288 Nov 16 18:11 scripts
lrwxrwxrwx 1 root root      35 Oct 18 00:19 security -> ../linux-headers-5.4.0-132/security
lrwxrwxrwx 1 root root      32 Oct 18 00:19 sound -> ../linux-headers-5.4.0-132/sound
drwxr-xr-x 3 root root    4096 Nov 16 18:11 tools
lrwxrwxrwx 1 root root      33 Oct 18 00:19 ubuntu -> ../linux-headers-5.4.0-132/ubuntu
lrwxrwxrwx 1 root root      30 Oct 18 00:19 usr -> ../linux-headers-5.4.0-132/usr
lrwxrwxrwx 1 root root      31 Oct 18 00:19 virt -> ../linux-headers-5.4.0-132/virt
`
The security, sound, ubuntu, usr and virt lines are in red while the rest are in blue.  Not sure what that means.  So I tried to reinstall openrazer.
`fungusgod@Cryptonomicon:/$ sudo apt update
Ign:1 http://packages.linuxmint.com una InRelease                                                                  
Hit:2 http://ppa.launchpad.net/appimagelauncher-team/stable/ubuntu focal InRelease                                 
Get:3 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                          
Hit:4 https://repo.nordvpn.com//deb/nordvpn/debian stable InRelease                                                
Hit:5 http://archive.ubuntu.com/ubuntu focal InRelease                                                             
Hit:6 https://download.docker.com/linux/debian buster InRelease                                                    
Hit:7 http://packages.linuxmint.com una Release                                                                    
Hit:8 http://archive.canonical.com/ubuntu focal InRelease                                                          
Get:9 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                              
Hit:10 http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease                   
Hit:12 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease
Get:13 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB] 
Hit:14 http://ppa.launchpad.net/wireshark-dev/stable/ubuntu focal InRelease           
Fetched 336 kB in 2s (151 kB/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
fungusgod@Cryptonomicon:/$ sudo dkms install openrazer-driver/3.5.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
KERNELDIR=/lib/modules/5.4.0-132-generic/build make driver...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.0-132-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/3.5.0/build/make.log for more information.
`
There is that make.log entry again.  This time it had events listed.
`DKMS make.log for openrazer-driver-3.5.0 for kernel 5.4.0-132-generic (x86_64)
Tue 22 Nov 2022 03:04:46 AM JST
-e
::^[[32m Compiling OpenRazer kernel modules^[[0m
========================================
make -C /lib/modules/5.4.0-132-generic/build M=/var/lib/dkms/openrazer-driver/3.5.0/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-132-generic'
Makefile:952: scripts/Makefile.ubsan: No such file or directory
make[1]: *** No rule to make target 'scripts/Makefile.ubsan'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-132-generic'
make: *** [Makefile:41: driver] Error 2
`
Now I am stuck again.

lah7, 2022-11-21:
>  I can't figure out why my post is formatting so weird.

Wrap the terminal output with triple backticks (```), I made edits to do that so it was easier to read.

---

I just updated a 20.04 VM to double check and it succeeded. I also noticed the output of `ls -l /usr/src/linux-headers-5.4.0-132-generic/` all had blue/aqua text colours, no red. I think red means that the symlink is broken, which could explain why it's not installing on your side.

I'd try reinstalling the kernel headers in case those files are just missing:

    sudo apt install --reinstall linux-headers-5.4.0-132 linux-headers-5.4.0-132-generic

Then try rebuilding via DKMS again. 🤞🏼

fungusgod, 2022-11-21:
https://github.com/lah7 Thank you Luke, that was exactly what it took.  Still learning linux, I tried to reinstall the headers but apparently had the wrong syntax.  After running your script, all blue entries and running 
```sudo dkms install openrazer-driver/3.5.0
Module openrazer-driver/3.5.0 already installed on kernel 5.4.0-132-generic/x86_64``` which is the last step before restarting the machine and OpenRazer.

closed 2022-11-21 by fungusgod