openrazer/openrazer #1962: Amdgpu, snapd failure after openrazer-meta package installation on Fedora 36

jm-boley, 2022-11-23:
### Distribution

Fedora 36
Kernel: 6.0.9-200.fc36.x86_64
Desktop: GNOME running Wayland + amdgpu

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I installed the openrazer-meta package from the razer repository. After restarting I found that my linux environment has broken in a couple of ways:
- The amdgpu and radeon kernel mods fail to load
- Snapd and dependent services fail to initialize

I've poked about a bit trying to find any useful information. Disclaimer: My superuser skills are a bit rusty. I first tried tackling the amdgpu problem. Running `lsmod | egrep '(radeon|amdgpu)'` returns an empty list, so those apparently aren't being loaded, and trying to `lsmod` them in fails. I've double-checked to make sure the system still recognizes the hardware, and executing `lspci -k | grep -A 3 -E "(VGA|3D)"` does show the gpu:
```
0a:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c1)
	Subsystem: Sapphire Technology Limited Device e448
0a:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller
	Subsystem: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller
```
The expected `Kernel driver in use` and `Kernel modules` fields are notably absent. I checked the kernel modules folder on the system with `find /lib/modules/$(uname -r) -type f -name '*.ko*' | grep '(gpu|radeon)'` and this is what I got back:
```
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/display/drm_display_helper.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/gud/gud.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/hyperv/hyperv_drm.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/qxl/qxl.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/scheduler/gpu-sched.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/solomon/ssd130x-i2c.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/solomon/ssd130x-spi.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/solomon/ssd130x.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/bochs.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/cirrus.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/gm12u320.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/ili9163.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/ili9486.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/tiny/panel-mipi-dbi.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/ttm/ttm.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/udl/udl.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/vboxvideo/vboxvideo.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/vgem/vgem.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/virtio/virtio-gpu.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/vkms/vkms.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/vmwgfx/vmwgfx.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/drm_buddy.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/drm_cma_helper.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/drm_mipi_dbi.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/drm_ttm_helper.ko.xz
/lib/modules/6.0.9-200.fc36.x86_64/kernel/drivers/gpu/drm/drm_vram_helper.ko.xz
```
So it seems that somehow the drm/amdgpu and radeon modules have completely gone missing.

Lastly, I checked the journald logs just to see what I'd find. I found a few entries that struck me as unusual, but this is where the rust really starts to show so I'm not sure how relevant most of this is. Using `journalctl -xb | grep gpu`:
```
Nov 23 10:12:39 fedora packagekitd[1561]: Failed to get cache filename for amd-gpu-firmware
Nov 23 10:12:39 fedora packagekitd[1561]: Failed to get cache filename for intel-gpu-firmware
Nov 23 10:12:39 fedora packagekitd[1561]: Failed to get cache filename for nvidia-gpu-firmware
Nov 23 10:14:50 fedora brave-browser.desktop[3304]: [3547:3547:1123/101450.077353:ERROR:gpu_memory_buffer_support_x11.cc(44)] dri3 extension not supported.
```

The second broken component is the snapd daemon. Querying systemctl with `sysctl list-units --failed` returns
```
  UNIT                                                  LOAD   ACTIVE SUB    DESCRIPTION                                    
● var-lib-snapd-snap-bare-5.mount                       loaded failed failed Mount unit for bare, revision 5
● var-lib-snapd-snap-core18-2620.mount                  loaded failed failed Mount unit for core18, revision 2620
● var-lib-snapd-snap-core18-2632.mount                  loaded failed failed Mount unit for core18, revision 2632
● var-lib-snapd-snap-core22-310.mount                   loaded failed failed Mount unit for core22, revision 310
● var-lib-snapd-snap-gnome\x2d3\x2d28\x2d1804-161.mount loaded failed failed Mount unit for gnome-3-28-1804, revision 161
● var-lib-snapd-snap-gnome\x2d3\x2d34\x2d1804-77.mount  loaded failed failed Mount unit for gnome-3-34-1804, revision 77
● var-lib-snapd-snap-gnome\x2d42\x2d2204-29.mount       loaded failed failed Mount unit for gnome-42-2204, revision 29
● var-lib-snapd-snap-gnome\x2d42\x2d2204-44.mount       loaded failed failed Mount unit for gnome-42-2204, revision 44
● var-lib-snapd-snap-gtk\x2dcommon\x2dthemes-1534.mount loaded failed failed Mount unit for gtk-common-themes, revision 1534
● var-lib-snapd-snap-gtk\x2dcommon\x2dthemes-1535.mount loaded failed failed Mount unit for gtk-common-themes, revision 1535
● var-lib-snapd-snap-slack-67.mount                     loaded failed failed Mount unit for slack, revision 67
● var-lib-snapd-snap-snapd-17336.mount                  loaded failed failed Mount unit for snapd, revision 17336
● var-lib-snapd-snap-snapd-17576.mount                  loaded failed failed Mount unit for snapd, revision 17576
● var-lib-snapd-snap-spotify-60.mount                   loaded failed failed Mount unit for spotify, revision 60
```
Unfortunately, the Snap documentation hosted by Canonical ([https://snapcraft.io/docs](https://snapcraft.io/docs)) appears to be somewhat lacking so I'm not sure exactly what to look for. If I had to guess, I'd say since it runs containerized apps it probably relies on a kernel mod which has either gone missing or some part of the kvm which is no longer loading properly; I am noticing some kvm-amd errors in the logs.

At any rate, both of these issues reared their ugly heads immediately after the install-restart, which I assume points to a problem (or problems) with the package install scripts.

I'm following up elsewhere with support requests as appropriate, however this seems like a pretty serious bug in the package install process so I'm guessing you'd want to know. Ironically, the openrazer driver works just fine.

z3ntu, 2022-11-24:
Have you tried uninstalling OpenRazer again? Does that solve the issue?

jm-boley, 2022-11-25:
I ended up going a slightly different route. With a little coaching from the guys on the Ask Fedora forum, I was able to reinstall the damaged OS components with `dnf upgrade --refresh` and `dnf distro-sync --refresh --allowerasing`, after disabling the razer-hardware repo. That seems to have mostly returned my system to the state it was in prior to installing the openrazer-meta package. As a bonus, I didn't even have to remove OpenRazer and it appears to be working.

Of course, that doesn't fix what appears to be an incredibly nasty bug in the openrazer-meta package install scripts, so that issue is still outstanding. As a support "ticket" this can be closed, but I'll let a moderator make that decision since that bug (I don't know what else it would be) is still out there.

z3ntu, 2022-11-25:
Since nobody else has reported this, I'm going to assume it was a problem isolated to your machine due to whatever reasons.
If this comes up again, we'll definitely look into it.

closed 2022-11-25 by z3ntu