openrazer/openrazer #1963: Laptop lid always active Razer Blade 2020 Advanced (Early 2020)

LexXxurio, 2022-11-23:
### Distribution

Arch Linux - Linux 6.0.9-arch1-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am trying to turn off the latop lid logo and used polychromatic as well as razergenie for that but after a reboot the logo is active again.
Even tried to stop the daemon and set the logo state in the persistence config file to false and restart the daemon but after each start of the daemon it is active again. 
Another issue might not be related to that is that I cannot configue the brightness nor the color of the logo. Only activating it is possible.

### Openrazer Version
```
openrazer-daemon --version
openrazer-daemon 3.5.0
```

### Device Info
```
lsusb | grep 1532
Bus 001 Device 004: ID 1532:0253 Razer USA, Ltd RZ09-0330, Gaming Laptop [Blade 15 Advanced (Early 2020)]
```

### Daemon Log
```
openrazer-daemon -rFv
Process 7903 stopped
Starting daemon.
2022-11-23 23:17:52 | razer                          | INFO     | Initialising Daemon (v3.5.0). Pid: 7965
2022-11-23 23:17:52 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeedWireless - (1532:0271)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3ProWired --------------- (1532:025A)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3TK --------------------- (1532:0A24)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2022-11-23 23:17:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
[... 184 lines cut ...]
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightRandom method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightSingle method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightDual method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | DBus call set_brightness
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlade15Advanced2020:BY2020A99500210, 'setBrightness', 3)
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | DBus call set_logo_active
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | DBus call set_static_effect
2022-11-23 23:17:52 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlade15Advanced2020:BY2020A99500210, 'setStatic', 0, 255, 0)
2022-11-23 23:17:52 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
[... cut ...]
2022-11-23 23:17:52 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2022-11-23 23:17:52 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2022-11-23 23:17:52 | razer                          | INFO     | Serving DBus
2022-11-23 23:18:02 | razer                          | DEBUG    | State recently changed, writing to disk
2022-11-23 23:18:02 | razer                          | DEBUG    | Writing persistence config
```

z3ntu, 2023-01-09:
I have some fixes for stuff like this in git master, could you try openrazer-git packages (Arch Linux)?

LexXxurio, 2023-01-11:
Nope, still the same. I used openrazer-meta-git but should be the same right?

z3ntu, 2023-01-11:
Yes `openrazer-meta-git` but you also need to have the same version of the other subpackages `openrazer-daemon-git` and `openrazer-driver-dkms` and `python-openrazer`. Just upgrading the meta package won't change anything.

LexXxurio, 2023-01-11:
Upgraded all to *-git packages and still the same issue.

z3ntu, 2023-01-15:
Thanks for trying, was worth a shot.

PR #2008 (2023-02-11): Fix persistence for lighting on/off state
The persistence file stored the active state for lighting as `True` or `False` and then converted these strings using `bool()`. This  will always give back true, causing lighting to be re-enabled whenever the daemon is restarted.

This should fix #1963, since I was having the same issue with my mouse.
closed 2023-02-11 by z3ntu