openrazer/openrazer #1965: TypeError during python-openrazer installation

didelphid, 2022-11-28:
### Distribution

OS: Arch Linux
Kernel: x86_64 Linux 6.0.10-zen2-1-zen

### Support

- [✓] Have you tried rebooting?
- [✓] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

All `openrazer` packages fail to install on Arch Linux Zen 6.0.10 with a TypeError on setup().

```python3 setup.py install --prefix=/usr --root=/home/maddox/.cache/yay/openrazer/pkg/python-openrazer
Traceback (most recent call last):
  File "/home/maddox/.cache/yay/openrazer/src/openrazer-3.5.0/pylib/setup.py", line 6, in <module>
    setup(
  File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 86, in setup
    _install_setup_requires(attrs)
  File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 75, in _install_setup_requires
    dist = MinimalDistribution(attrs)
  File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 57, in __init__
    super().__init__(filtered)
  File "/usr/lib/python3.10/site-packages/setuptools/dist.py", line 473, in __init__
    for ep in metadata.entry_points(group='distutils.setup_keywords'):
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1009, in entry_points
    return SelectableGroups.load(eps).select(**params)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 459, in load
    ordered = sorted(eps, key=by_group)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1006, in <genexpr>
    eps = itertools.chain.from_iterable(
  File "/usr/lib/python3.10/importlib/metadata/_itertools.py", line 16, in unique_everseen
    k = key(element)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 941, in _normalized_name
    return self._name_from_stem(stem) or super()._normalized_name
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 622, in _normalized_name
    return Prepared.normalize(self.name)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 871, in normalize
    return re.sub(r"[-_.]+", "-", name).lower().replace('-', '_')
  File "/usr/lib/python3.10/re.py", line 209, in sub
    return _compile(pattern, flags).sub(repl, string, count)
TypeError: expected string or bytes-like object
make[1]: *** [Makefile:9: install] Error 1
make: *** [Makefile:91: python_library_install] Error 2
==> ERROR: A failure occurred in package_python-openrazer().
    Aborting...```

z3ntu, 2022-11-28:
Can you try building in a chroot, e.g. with `extra-x86_64-build` ? https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot

I haven't seen this error before and looking at the stacktrace, something might just be broken on your machine.

didelphid, 2022-12-02:
Hi there, sorry for the delay! This turned out to be the issue. openrazer is now looking and running great, and I'll look into fixing my machine soon. Thank you for the suggestion!

closed 2022-12-02 by didelphid