openrazer/openrazer #1966: Support for Razer Naga V2 Hyperspeed

sam-goode, 2022-11-29:
### Device Info

- Name: Razer Naga V2 Hyperspeed
- Product Number: RZ01-03600100
- Model Number: RZ01-0360

### Device Information
```Bus 001 Device 007: ID 1532:00b4 Razer USA, Ltd Razer Naga V2 HyperSpeed
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b4 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga V2 HyperSpeed
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      97
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```total 0
drwxr-xr-x 2 root root 300 Nov 29 17:54 .
drwxr-xr-x 4 root root 800 Nov 29 19:13 ..
lrwxrwxrwx 1 root root  10 Nov 29 17:54 usb-046d_Logitech_StreamCam_F0023F65-event-if00 -> ../event13
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-04d9_USB-HID_Keyboard-event-if02 -> ../event5
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-04d9_USB-HID_Keyboard-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-04d9_USB-HID_Keyboard-if02-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-if01-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-if02-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Razer_Razer_Naga_V2_HyperSpeed_000000000000-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Valve_Software_Steam_Controller-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Nov 29 17:54 usb-Valve_Software_Steam_Controller-mouse -> ../mouse0
```

bertrand-benoit, 2023-01-24:
Hello, any ETA on addition of support for this device?

bertrand-benoit, 2023-01-24:
I think it will be the same, but just in case, I put information too (and there are missing information in previous post, you will find them at end of this one):

```
Bus 007 Device 004: ID 1532:00b4 Razer USA, Ltd Razer Naga V2 HyperSpeed
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b4 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga V2 HyperSpeed
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      97
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
[... 78 lines cut ...]
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

007:004:000:DESCRIPTOR         1674569299.074039
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 06 00 FF 05 0C 0A 38 02 15 81 25 7F 75 08 95 01
 81 06 05 01 09 38 15 81 25 7F 75 08 95 01 81 06
 09 30 09 31 16 00 80 26 FF 7F 75 10 95 02 81 06
 C0 06 00 FF 09 02 15 00 25 01 75 08 95 5A B1 01
 C0
```

RobotRoss, 2023-03-02:
Just want to add my +1 to this. Just got the device, and it's pretty good but the lack of support here is a bummer!

rhaamo, 2023-03-13:
I have quickly hacked something and I can get battery reporting on daemon logs.
But when launching polychromatic it yells at me about device not having matrix so I am probably missing something, RazerGenie does show the right DPI stage when switching on the mouse and relaunching the app, hasn't really fiddled with anything else.
Just put some define here and there based on the basilisk v3 pro and the naga pro. 
That was using the wireless receiver, I don't really know for the bluetooth part...

Anyway if a dev wants I can provide pcaps, more infos, and do some testing.

commit 1102c792 (2023-10-24): Add support for Razer Naga V2 HyperSpeed (Receiver)

Resolves: #1966
PR #2141 (2023-10-24): Add support for Razer Naga V2 HyperSpeed (Receiver)

Resolves: #1966

Copied alot from PR #2035 and fixed that dpi was set using openrazer python api

Is there anything else I should test/verify?
closed 2023-10-24 by z3ntu