openrazer/openrazer #1968: openrazer-daemon.service error

Dissonanc3, 2022-11-30:
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/etc/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-11-30 11:53:56 CET; 23s ago
       Docs: man:openrazer-daemon(8)
    Process: 15438 ExecStart=/nix/store/w52syhqx13vx3krxcwnhdx4lh4riba06-openrazer_daemon-3.1.0/bin/openrazer-daemon --config /nix/store/hkyz2hga0n5n3ppnal9v97q0ih2hvzaz-razer.conf --foreground (code=exited, status=1/FAILURE)
   Main PID: 15438 (code=exited, status=1/FAILURE)
        CPU: 91ms

11月 30 11:53:56 nixos systemd[1302]: openrazer-daemon.service: Scheduled restart job, restart counter is at 5.
11月 30 11:53:56 nixos systemd[1302]: Stopped Daemon to manage razer devices in userspace.
11月 30 11:53:56 nixos systemd[1302]: openrazer-daemon.service: Start request repeated too quickly.
11月 30 11:53:56 nixos systemd[1302]: openrazer-daemon.service: Failed with result 'exit-code'.
11月 30 11:53:56 nixos systemd[1302]: Failed to start Daemon to manage razer devices in userspace.

z3ntu, 2022-11-30:
Check `journalctl -eu openrazer-daemon.service` if there's any error

Dissonanc3, 2022-11-30:
I get no entries

z3ntu, 2022-11-30:
I don't know much about NixOS but try running `/nix/store/w52syhqx13vx3krxcwnhdx4lh4riba06-openrazer_daemon-3.1.0/bin/openrazer-daemon --config /nix/store/hkyz2hga0n5n3ppnal9v97q0ih2hvzaz-razer.conf --foreground` and see if that throws an error.

Dissonanc3, 2022-11-30:
2022-11-30 12:04:41 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 16960
2022-11-30 12:04:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-11-30 12:04:41 | razer                          | INFO     | Found device.0: 0003:1532:0078.0001
2022-11-30 12:04:41 | razer.device0                  | INFO     | Initialising device.0 RazerViper
2022-11-30 12:04:41 | razer                          | INFO     | Serving DBus

Dissonanc3, 2022-11-30:
oh it works now

Dissonanc3, 2022-11-30:
thank you very much for help

closed 2022-11-30 by Dissonanc3

z3ntu, 2022-11-30:
If the systemd service doesn't work, then I believe it's a bug in the NixOS packaging so please ask there for help.