openrazer/openrazer #1970: razer kraken v3 hypersense

10ked, 2022-12-04:
Please post the output of
    lsusb -d 1532: -v
Bus 001 Device 007: ID 1532:0533 Razer USA, Ltd Razer Kraken V3 HyperSense
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0533 
  bcdDevice            0.04
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken V3 HyperSense
  iSerial                 3 000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00f7
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004f
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 2
        bSourceID               1
        bControlSize            1
        bmaControls(0)       0x01
[... 285 lines cut ...]
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
 
basically i am getting initial support here and can access my headphone and mike in kde plasma
system settings and can hear it working when i toggle the volume control on both the headphones
and the sound kmixer but with unstable very basey sound and unstable unusable mike and when i access the mike
in steam i get feedback but some voice when i test the mike in big picture. No doubt there are probably some
caveats and howtos on usb headphone i have not seen and i WILL be looking for more help so this is only 
an initial entry so please bear with me. I am presently using slackware 15 64 bit multilib .

![razer_kraken_v3_hypersense](https://user-images.githubusercontent.com/9946347/205467959-ae2c2fae-74f7-4128-9143-a68920ab2c01.png)

10ked, 2022-12-04:
ok i just found out that in the configuration of pavucontrol if all the profiles of the other audio devices are disabled
the usb headset then operates much more stable then it did before as in the mike now works and the sound
is much improved although finer support in openrazer for the headset and more complete control of it's
features like it's led lights and their features are not present as of yet so this thread is not closed as of yet.

Rumir, 2025-01-27:
I've seen that this request is still open.

I have a .pcapng-file with usb capture where I changed the lighting in a Win11 VM. The data has been captures on my Linux host.

Do I have to do something more with the file, or just post it here?

z3ntu, 2025-02-15:
@Rumir Yes, just post it here. I think GitHub doesn't like you uploading a file with the .pcapng extension, so just pack it into a .zip or change the file extension if it doesn't work.

Rumir, 2025-02-15:
@z3ntu I packed the filtered capture file into a .zip: [razer_kraken_v3_filtered.zip](https://github.com/user-attachments/files/18810067/razer_kraken_v3_filtered.zip)

ElectronicIV, 2026-11-07:
I've opened a PR adding lighting support for this headset: #2981 

I've got Static, breathing, spectrum, custom colour, off and brightness working through the daemon and Polychromatic.
Haptics aren't covered yet.
You may test it if you want.

Thanks @Rumir for your capture!