openrazer/openrazer #1972: Cannot Update; Throwing Error "No module named 'distutils.command.bdist_wininst' "

AgentLoneStar007, 2022-12-05:
### Distribution
EndeavourOS(Arch-based)

Kernel version 6.0.10-arch2-1

### Support

- [ ] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi. I simply tried to update everything this morning on my PC using `yay -Syu`, and everything went well until it tried to compile the OpenRazer Daemon, which resulted in a Python error of "`No module named 'distutils.command.bdist_wininst'`".
Full output log:
```
:: Installing OpenRazer python library
=====================================================
python3 setup.py install --prefix=/usr --root=/home/agent/.cache/yay/openrazer/pkg/python-openrazer
/usr/lib/python3.10/site-packages/setuptools_scm/integration.py:27: RuntimeWarning: 
ERROR: setuptools==44.1.1 is used in combination with setuptools_scm>=6.x

Your build configuration is incomplete and previously worked by accident!
setuptools_scm requires setuptools>=45

This happens as setuptools is unable to replace itself when a activated build dependency
requires a more recent setuptools version
(it does not respect "setuptools>X" in setup_requires).

setuptools>=31 is required for setup.cfg metadata support
setuptools>=42 is required for pyproject.toml configuration support

Suggested workarounds if applicable:
 - preinstalling build dependencies like setuptools_scm before running setup.py
 - installing setuptools_scm using the system package manager to ensure consistency
 - migrating from the deprecated setup_requires mechanism to pep517/518
   and using a pyproject.toml to declare build dependencies
   which are reliably pre-installed before running the build tools

  warnings.warn(
running install
running build
running build_py
creating build
creating build/lib
creating build/lib/openrazer
copying openrazer/__init__.py -> build/lib/openrazer
creating build/lib/openrazer/client
copying openrazer/client/macro.py -> build/lib/openrazer/client
copying openrazer/client/device.py -> build/lib/openrazer/client
copying openrazer/client/fx.py -> build/lib/openrazer/client
copying openrazer/client/__init__.py -> build/lib/openrazer/client
copying openrazer/client/constants.py -> build/lib/openrazer/client
copying openrazer/client/debug.py -> build/lib/openrazer/client
creating build/lib/openrazer/client/devices
copying openrazer/client/devices/mousemat.py -> build/lib/openrazer/client/devices
copying openrazer/client/devices/__init__.py -> build/lib/openrazer/client/devices
copying openrazer/client/devices/mice.py -> build/lib/openrazer/client/devices
copying openrazer/client/devices/keyboard.py -> build/lib/openrazer/client/devices
running install_lib
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/macro.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/device.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/fx.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/__init__.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/constants.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
copying build/lib/openrazer/client/debug.py -> /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client
creating /home/agent/.cache/yay/openrazer/pkg/python-openrazer/usr/lib/python3.10/site-packages/openrazer/client/devices
[... 42 lines cut ...]
    self.run_command(cmd_name)
  File "/usr/lib/python3.10/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/agent/.local/lib/python3.10/site-packages/setuptools/command/install_scripts.py", line 35, in run
    bw_cmd = self.get_finalized_command("bdist_wininst")
  File "/usr/lib/python3.10/distutils/cmd.py", line 298, in get_finalized_command
    cmd_obj = self.distribution.get_command_obj(command, create)
  File "/usr/lib/python3.10/distutils/dist.py", line 857, in get_command_obj
    klass = self.get_command_class(command)
  File "/home/agent/.local/lib/python3.10/site-packages/setuptools/dist.py", line 834, in get_command_class
    self.cmdclass[command] = cmdclass = ep.load()
  File "/home/agent/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/home/agent/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/agent/.local/lib/python3.10/site-packages/setuptools/command/bdist_wininst.py", line 1, in <module>
    import distutils.command.bdist_wininst as orig
ModuleNotFoundError: No module named 'distutils.command.bdist_wininst'
make[1]: *** [Makefile:9: install] Error 1
make: *** [Makefile:91: python_library_install] Error 2
==> ERROR: A failure occurred in package_python-openrazer().
    Aborting...
```

z3ntu, 2022-12-05:
Try building in a clean chroot, e.g. using `extra-x86_64-build` https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot

z3ntu, 2022-12-05:
Also Arch repos contain setuptools v65, not v44 as you seem to have installed according to the error message.
```
$ pacman -Q python-setuptools
python-setuptools 1:65.6.3-1
```
Are you running something like `sudo pip install` sometimes?

AgentLoneStar007, 2022-12-05:
I've never ran pip in sudo before.

AgentLoneStar007, 2022-12-06:
Alright, so after using a the devtools script in the package directory, it gave this error at the end:
```
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
installing pyalpm...
installing licenses...
installing elfutils...
installing python-pyelftools...
installing namcap...
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
Checking PKGBUILD
Checking openrazer-daemon-3.5.1-1-any.pkg.tar.zst
openrazer-daemon W: Dependency python-dbus included but already satisfied
Checking openrazer-driver-dkms-3.5.1-1-any.pkg.tar.zst
openrazer-driver-dkms W: Dependency included and not needed ('dkms')
openrazer-driver-dkms W: Dependency included and not needed ('udev')
Checking openrazer-meta-3.5.1-1-any.pkg.tar.zst
openrazer-meta W: Dependency included and not needed ('openrazer-driver-dkms')
openrazer-meta W: Dependency included and not needed ('openrazer-daemon')
openrazer-meta W: Dependency included and not needed ('python-openrazer')
Checking python-openrazer-3.5.1-1-any.pkg.tar.zst
==> Running checkpkg
error: target not found: python-openrazer
error: target not found: openrazer-daemon
error: target not found: openrazer-driver-dkms
error: target not found: openrazer-meta
==> WARNING: Skipped checkpkg due to missing repo packages
```
[Here's the full output.](https://pastebin.com/TLiGYZB2) (It's a bit long. Output of installing Openrazer and the build script.)

AgentLoneStar007, 2022-12-07:
Well, that worked. Thanks. It's funny, though, because _as_ I go to update to check, Protontricks is doing the exact same thing, and what I did for this didn't work on that.

Speaking of which, if anyone has this issue and needs help, here's what I did:

- Install *devtools* from the Arch repository
- Go to *~/.cache/yay/openrazer*
- Run the command *extra-x86_64-build* in that directory
- When finished, install all packages ending in *.pkg.tar.zst* with *sudo pacman -U (packagename)*