openrazer/openrazer #1974: Lighting control really poor

Karnor, 2022-12-06:
### Distribution

Kubuntu 22.04

### Support

- [ ] Have you tried rebooting? Yes multiple times
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

I have the following devices
Razer Blackwidow v3
Razer DeathAdder v2

The various lighting effects are very poor.

On both devices the Breath mode leaves something like a 1 second delay between colour pulses leaving dark devices for far too long which is not good in a dark room. Even as a touch typist it's just not good.

On BW the Ripple mode turns all the lights off.

On BW Wave mode really needs a way to control the wave speed it's reminiscent of speeding cars at night, just too fast.

So effectively as far as a lighting spectacle goes I'm restricted to using just Static. and Spectrum, though in truth Spectrum could do with a speed control because there are times the transition between colours seems a bit too slow.

z3ntu, 2022-12-06:
Breath mode is provided by the device firmware, so blame Razer there.

Ripple mode is expected to have all leds off by default and when you touch a keyboard button then a ripple appears. Does this not work for you?

Wave seems to in theory have a speed parameter but this is not configurable in OpenRazer and hardcoded in the driver. Here you could try adjusting `razer_chroma_extended_matrix_effect_wave` the `report.arguments[4]` (in `driver/razerchromacommon.c`). For testing these changes you can use https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md#build-the-kernel-module (also maybe read the sections above that)

Karnor, 2022-12-08:
No Ripple mode does not work for me.

When I switch to ripple mode the keyboard goes dark and does not light up when I touch any keys.

z3ntu, 2022-12-08:
Okay, ripple should be working.

What's the output of this? `ls -lah /dev/input/by-id/`

Karnor, 2022-12-08:
drwxr-xr-x 2 root root 300 Dec  8 12:22 .
drwxr-xr-x 4 root root 740 Dec  8 12:23 ..
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Plantronics_RIG_800HD-event-if03 -> ../event13
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_BlackWidow_V3-event-if04 -> ../event25
lrwxrwxrwx 1 root root   9 Dec  8 12:22 usb-Razer_Razer_BlackWidow_V3-event-kbd -> ../event3
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_BlackWidow_V3-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_BlackWidow_V3-if02-event-mouse -> ../event19
lrwxrwxrwx 1 root root   9 Dec  8 12:22 usb-Razer_Razer_BlackWidow_V3-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_DeathAdder_V2-event-mouse -> ../event10
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_DeathAdder_V2-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_DeathAdder_V2-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 Dec  8 12:22 usb-Razer_Razer_DeathAdder_V2-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Dec  8 12:22 usb-Razer_Razer_Nostromo-event-kbd -> ../event23
lrwxrwxrwx 1 root root   9 Dec  8 12:22 usb-Razer_Razer_Nostromo-if01-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Dec  8 12:22 usb-Razer_Razer_Nostromo-if01-mouse -> ../mouse1

z3ntu, 2022-12-08:
Can you stop the daemon `systemctl --user stop openrazer-daemon.service` and then run `openrazer-daemon -Fv` ? 

You can also try uncommenting this line in your install https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/key_event_management.py#L385 and see if any log lines appear from this when pressing buttons on the keyboard when in ripple mode. But please know that this will (obviously) log the keys you are using to where the daemon logs to, so you should disable that asap again.

Karnor, 2022-12-11:
openrazer-daemon -Fv output
Starting daemon.
2022-12-11 16:02:40 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 25021
2022-12-11 16:02:40 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeedWireless - (1532:0271)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3ProWired --------------- (1532:025A)
2022-12-11 16:02:40 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3TK --------------------- (1532:0A24)
[... cut ...]
2022-12-11 16:02:40 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2022-12-11 16:02:40 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2022-12-11 16:02:40 | razer                          | INFO     | Serving DBus
2022-12-11 16:02:50 | razer                          | DEBUG    | State recently changed, writing to disk
2022-12-11 16:02:50 | razer                          | DEBUG    | Writing persistence config

z3ntu, 2022-12-11:
Okay, let's try another way. For the different `usb-Razer_Razer_BlackWidow_V3-*` files in `/dev/input/by-id/` run `cat` on one, press some keys on the keyboard and see if you get some binary data out of it.
E.g. for me it looks like this
```
$ cat /dev/input/by-id/usb-Razer_Razer_BlackWidow_Chroma-event-kbd
^C
$ cat /dev/input/by-id/usb-Razer_Razer_BlackWidow_Chroma-if01-event-kbd
B�c՝
    XB�c՝
         `B�c՝
              C�cx'�C�cx'C�cx'C�c��
�C�c��
C�c��
C�c>�
     �C�c>�
           8C�c>�
D�cW2�D�cW28D�cW2D�c],�D�c],D�c],D�c���D�c��8D�c��E�cN�E�cNE�cNE�c���E�c��E�c��E�c�3�E�c�3E�c�3E�c���E�c��E�c��E�c���E�c��E�c��E�c�+�E�c�+E�c�+E�c}
                                                                                                                                                   �E�c}
                                                                                                                                                        E�c}
                                                                                                                                                            E�c��
    E�c��
         .E�c��
               ^C

```
And tell me which of the files produces output.

Karnor, 2022-12-12:
> You can also try uncommenting this line in your install https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/key_event_management.py#L385 and see if any log lines appear [...]

Tried that then realized I'm not entirely sure where the daemon logs to so don't know if any keys were actually logged.

If it's the same log as the openrazer-daemon -Fv output, which I did try, then no keys were logged. If it's a different log I should be looking at, please let me know.

> Okay, let's try another way. For the different `usb-Razer_Razer_BlackWidow_V3-*` files in `/dev/input/by-id/` run `cat` on one, press some keys on the keyboard and see if you get some binary data [...]

I have

- **/dev/input/by-id/usb-Razer_Razer_BlackWidow_V3-event-kbd** which produces no output
- **/dev/input/by-id/usb-Razer_Razer_BlackWidow_V3-if01-event-kbd** which does produce output.

z3ntu, 2022-12-13:
Interesting, I don't think it'll help but can you change https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/keyboards.py#L849 to be `EVENT_FILE_REGEX = re.compile(r'.*BlackWidow_V3-if01-event-kbd')` ? You find that file somewhere in /usr/lib/python3*/*-packages/openrazer_daemon I think, you can edit it there in-place

closed 2023-05-14 by github-actions[bot]