openrazer/openrazer #1976: Only legacy macros available due to no access to uinput

Karnor, 2022-12-08:
### Distribution

kubuntu 22.04

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When I open snake I have a warning message saying

`Snake supports 3 different Macro systems, 2 of them require access to /dev/uinput, which is not  currently available, so only Legacy OpenRazer macros are available. See Project Wiki for further explanation.`

But unless I'm blind, the wiki has no further information.

/dev/uinput definitely exists
`crw-rw----+ 1 root root 10, 223 Dec  8 12:22 /dev/uinput`

How do I fix this so I have access to the later macro systems?

z3ntu, 2022-12-08:
The warning is from Snake UI, see there then I guess: https://github.com/bithatch/snake/wiki/The-3-Macro-Systems

closed 2022-12-08 by z3ntu

FinalBosses, 2022-12-29:
For me, I only needed to run `sudo apt install bamfdaemon` to fix the issue. Everytime it launched up, it would give me an error about bamf, but clicking the retry button worked. Then I would get an error about /dev/uinput at the top of my screen. It turns out that the macro systems use bamf and need bamfdaemon to work. I don't know why it was mentioning /dev/uinput when that had nothing to do with it. Let me know if this solves your issue.