openrazer/openrazer #1977: Razer Viper Ultimate - hundreds of "razermouse: Command timed out" in journalctl

HanM23, 2022-12-08:
### Distribution

Manjaro 22 - XFCE

### Support

- [ yes] Have you tried rebooting?
- [ yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi, since the last update i have hundreds of "razermouse: Command timed out" messages in my journalctl when my mouse is in idle state when the screensaver is on.
This evening my computer was on with the screensaver and i have this message twice exactly every 1 minute. I guess there is one for the mouse and one for the dock, see :

```
déc. 08 21:51:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:51:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:50:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:50:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:49:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:49:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:49:14 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:48:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:48:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:47:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:47:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:46:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:46:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:45:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:45:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:44:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:44:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:43:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:43:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:42:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:42:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:41:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:41:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:40:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:40:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:39:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:39:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:39:13 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:38:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:38:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:37:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:37:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
déc. 08 21:36:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 84 Params: 00000000000000000000000000000000 .
déc. 08 21:36:48 <username> kernel: razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .

```

I have hundreds like this. My `/home/<username>/.local/share/openrazer/logs/razer.log` file is normal to me, there is not something unusual.
Am i alone with those messages ?
Thanks for your help.

z3ntu, 2022-12-09:
Is this when using the mouse in wireless mode? I imagine it switches to some energy saving mode when not used and therefore either doesn't respond to requests at all or just very slowly. You could try increasing the timeout in razermouse_driver.c to see if that changes anything. The 1 packet per minute is probably checking battery level, but not sure right now what the frequency of that is.

HanM23, 2022-12-09:
> Is this when using the mouse in wireless mode? I imagine it switches to some energy saving mode when not used and therefore either doesn't respond to requests at all or just very slowly. You could [...]

Yes, you are right, this happens on wireless mode only, when docked, i do not have these messages.
I had a look into `razermouse_driver.c` and, there is `RAZER_CMD_TIMEOUT`, but i really do not know what to do with it. My knowledge with C is very limited :(
Do you have any idea ?
Thank you

z3ntu, 2022-12-09:
Change these defines https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.h#L102-L103
I'd try adding a zero to both values

They're getting used here btw https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L68-L73

closed 2023-05-09 by github-actions[bot]

spikespaz, 2024-02-03:
Is this the same as #2200?

HanM23, 2024-02-03:
> Is this the same as #2200?

Potentially yes, i have those messages in journalctl each time my mouse goes idle

gudjonbe, 2025-01-30:
I have the same issue