openrazer/openrazer #1980: DeathStalker V2 volume wheel acts like a mouse wheel

simplyroman, 2022-12-10:
### Distribution

Manjaro 21 KDE Plasma

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The volume wheel acts like a mouse wheel with openrazer-driver_3.5.1.

A possible solution could be [the same problem as with Ornata Chroma V2](https://github.com/openrazer/openrazer/commit/5845e9d6fc14225ee3a9f0cfe33bec734ca3caa6

I don't have the programming skills to test it by myself. I hope someone has the possibility.

simplyroman, 2022-12-10:
@AZeroEight Thanks for bringing the support for the keyboard. Are you facing the same problem with the keyboard?

AZeroEight, 2022-12-10:
Sometimes, it works correctly and sometimes like a mouse wheel. I have not been able to pinpoint the cause yet.

AZeroEight, 2022-12-10:
> Do you mind giving the possible fix a chance?

I will. But it won't be before monday.

AZeroEight, 2022-12-14:
Hi @simplyroman,

I have not yet found a way to reproduce the issue. In my current setup the volume wheel acts correctly (openrazer 3.5.1, Kernel 5.15.80) and I did not manage to make it misbehave. If you find a way, please let me know but I cannot implement a fix for an issue I don't have.

simplyroman, 2022-12-14:
@AZeroEight I would appreciate it. Changing the modules sounds not too difficult.

simplyroman, 2022-12-15:
@AZeroEight I found a way to remap the keys. A fix is no more needed for me. I used [input-remapper](https://github.com/sezanzeb/input-remapper). I keep the issue open so someone who googled a solution can see this.

AZeroEight, 2022-12-15:
Hi @simplyroman,

using input-remapper is just a workaround in my opinion. I would have the module ready - you just need to copy it to your kernel directory - I presume you are using Kernel 5.15.81-1-MANJARO?

You need to unload the existing module:
```shell
rmmod razerkbd
```

And load the new one:
```shell
insmod ./razerkbd.ko
```

let me know if this solves the issue for and I will create a pull request, so hopefully with the next revision this is part of history.

simplyroman, 2022-12-15:
@AZeroEight When I want to load the new module (sudo insmod ./rtazerkbd.ko) this happens:
`insmod: ERROR: could not insert module ./razerkbd.ko: Invalid module format`

AZeroEight, 2022-12-16:
@simplyroman please find the module for kernel 6.0.11-1-MANJARO attached.

[razerkbd.ko.gz](https://github.com/openrazer/openrazer/files/10243781/razerkbd.ko.gz)

simplyroman, 2022-12-16:
Hi @AZeroEight, I've tested it and can confirm that it fixes the issue. Thank you very much.

AZeroEight, 2022-12-16:
I've created the pull request so hopefully the fix will be in the next release.

PR #1983 (2022-12-20): Fixed Volume wheel acting like mouse wheel for Razer DeathStalker V2
Resolves #1980
closed 2022-12-20 by z3ntu