openrazer/openrazer #1981: Use $releasever yum variable for Fedora repo instead of fixed Fedora version number

thunderysteak, 2022-12-11:
### Distribution

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Using a fixed version when supporting multiple versions of Fedora in the dot repo file will cause breakages during Fedora version updates, and should be following best practices of using the $releasever yum variable in the repo file instead.

I am submitting an Issue instead of a pull request as I can't locate the files used with OpenSuse build service in this repo.

How `hardware:razer.repo` would look like after the changes
```
[hardware_razer]
name=hardware:razer (Fedora_$releasever)
type=rpm-md
baseurl=https://download.opensuse.org/repositories/hardware:/razer/Fedora_$releasever/
gpgcheck=1
gpgkey=https://download.opensuse.org/repositories/hardware:/razer/Fedora_$releasever/repodata/repomd.xml.key
enabled=1
```

Local testing with the modified repo file:
```
[thunderysteak@steak-fedora-blade ~]$ sudo nano /etc/yum.repos.d/hardware\:razer.repo 
[sudo] password for thunderysteak: 
[thunderysteak@steak-fedora-blade ~]$ sudo dnf install openrazer-meta
hardware:razer (Fedora_37)                                                                                                                                                                                    12 kB/s | 8.1 kB     00:00    
Dependencies resolved.
=============================================================================================================================================================================================================================================
 Package                                                               Architecture                                   Version                                                   Repository                                              Size
=============================================================================================================================================================================================================================================
Installing:
 openrazer-meta                                                        noarch                                         3.5.1-1.1                                                 hardware_razer                                         6.9 k
Installing dependencies:
 dkms                                                                  noarch                                         3.0.8-1.fc37                                              updates                                                 87 k
 openrazer-daemon                                                      noarch                                         3.5.1-1.1                                                 hardware_razer                                         191 k
 openrazer-kernel-modules-dkms                                         noarch                                         3.5.1-1.1                                                 hardware_razer                                          61 k
 python3-daemonize                                                     noarch                                         2.5.0-14.fc37                                             fedora                                                  18 k
 python3-numpy                                                         x86_64                                         1:1.22.0-7.fc37                                           fedora                                                 7.3 M
 python3-openrazer                                                     noarch                                         3.5.1-1.1                                                 hardware_razer                                          60 k
 xautomation                                                           x86_64                                         1.09-7.fc37                                               fedora                                                  61 k

Transaction Summary
=============================================================================================================================================================================================================================================
Install  8 Packages

Total download size: 7.8 M
Installed size: 47 M
Is this ok [y/N]: 
```

MrAn0nym, 2024-11-02:
@z3ntu Can you reopen this? @thunderysteak's suggestion is an actual fix for #2187. At the moment the repofile looks something like this:
```
[hardware_razer]
name=hardware:razer (Fedora_40)
type=rpm-md
baseurl=https://download.opensuse.org/repositories/hardware:/razer/Fedora_40/
gpgcheck=1
gpgkey=https://download.opensuse.org/repositories/hardware:/razer/Fedora_40/repodata/repomd.xml.key
enabled=1
```
Note that "Fedora_40" is hardcoded. That's why the system upgrade isn't able to fetch from the Fedora_41 repo. The official way of handling this is using `Fedora_$releasever` instead.

z3ntu, 2024-11-05:
@MrAn0nym Right now the .repo file is being fetched from download.opensuse.org which I have no direct control over. 

However we can host our own, made a draft here: https://public.lucaweiss.eu/hardware:razer.repo

```
sudo dnf config-manager addrepo --from-repofile=https://public.lucaweiss.eu/hardware:razer.repo
```

Could you try if this works as expected?

MrAn0nym, 2024-11-05:
@z3ntu Installation from using this repofile works like a charm.

I wasn't able to test system upgrades, but I'm fairly positive that this should fix # 2187 due to following test:

```
sudo dnf install openrazer-meta --releasever=42
```
Using the opensuse's repofile allows me to install `openrazer-meta`, even though a build for fedora 42 does not exist yet ==> DNF must have downloaded version 41.
Doing the same with your draft it tells me, that the package wasn't found, just as it should.

closed 2024-11-07 by lah7