openrazer/openrazer #1982: Support for Razer Wolverine V2 Chroma

ZiViZiViZ, 2022-12-14:
### Device
[Razer Wolverine V2 Chroma](https://www.razer.com/console-controllers/razer-wolverine-v2)

### Distribution
Arch Linux

### Support

- [x] Tried rebooting
- [x] Looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page

### Issue
The device is recognized and works out of the box, however, I'd like to configure the buttons as I don't their current configuration and for that I need the [Razer Controller Setup For Xbox](https://www.microsoft.com/en-us/p/razer-controller-setup-for-xbox/9nsxxh05pfrq) but I don't have a windows machine:

[image]

z3ntu, 2022-12-20:
For this button mapping you should be able to use https://github.com/sezanzeb/input-remapper

Does this device have any RGB lights or other non-button binding settings that can be adjusted on Windows? If not there's nothing really to support from OpenRazer side.

ZiViZiViZ, 2022-12-25:
Thank you @z3mtu I'll try [Input Remapper](https://github.com/sezanzeb/input-remapper)

Yes, it does have "RGB lights or other non-button binding settings that can be adjusted on Windows": http://youtu.be/iHsguNSUewM&t=347s

z3ntu, 2023-01-09:
Please provide USB packet captures of the changing the settings then: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

BOSSoNe0013, 2023-01-13:
Hi @ZiViZiViZ ,
I've two Wolverine V2 Chroma which work fine on my system but they are not detected by Razer Controller Setup when i run the app in a virtual host (i've tried with VirtualBox and KVM) so i can't snif the usb communication.
Have you been able to do it successfully ?

ZiViZiViZ, 2023-01-13:
Hi @z3ntu @BOSSoNe0013
Unfortunately, I will not be able to provide USB packet captures anymore.
Since I had to return the controller while I was able to.  I don't want to pay for a premium product unless I can use most of it's functionalities.

ZiViZiViZ, 2023-01-13:
Should I close the issue?

BOSSoNe0013, 2023-01-13:
Do not close the issue @ZiViZiViZ , i'll certainly found a solution to capture usb communication

GioPanaro, 2023-06-14:
Is there any way I can help with this issue? I own one of these and would love for it to be supported.

diztorded, 2023-07-10:
I also have the Razer Wolverine V2 and am willing to do any testing or capturing. Let me know what needs done and how I can help.

z3ntu, 2023-07-16:
Still waiting for usb packet captures I believe https://github.com/openrazer/openrazer/issues/1982?notification_referrer_id=NT_kwDOADmAtLI1MDg3NzQxODIwOjM3Njg1MDA&notifications_query=repo%3Aopenrazer%2Fopenrazer#issuecomment-1376023663

diztorded, 2023-07-21:
Installing Win10 now. Will start working on a capture.

diztorded, 2023-07-21:
Just realized I do not have the chroma version, so I could get buttons, but not lights. Sorry I cannot help.

sicausier, 2023-09-27:
Having never done anything like this i "think" i have captured packets, what should i de with the pcapng file?

sicausier, 2023-09-27:
https://drive.google.com/file/d/16lZWG8bwaVu6s4Yj1_HOrUWNRuRlqJG0/view?usp=sharing

I'll leave this here in case that's all you need

z3ntu, 2023-10-14:
While I do see some packets that normally would be normal Razer packets, nothing in there looks like something used on other devices, so it's definitely not a trivial job to add support for this device.

Also can anybody post `lsusb | grep 1532` output with the controller connected? I don't think the usb pid is posted anywhere in this issue yet.

BOSSoNe0013, 2023-10-14:
Bus 001 Device 028: ID 1532:0a2e Razer USA, Ltd Razer Wolverine V2 Chroma

sicausier, 2023-10-15:
could i have messed something up with the capture i can run it again if so?

z3ntu, 2023-10-16:
Possible but it doesn't sound unlikely that they're using a completely different way to control the lights etc on this device.

RamboN117, 2024-05-26:
Hi, I've got the same device. Is there anything I could do to help?

p.s I'm new to linux/openrazer, I'll probably ask questions including the above

Sooyka, 2024-06-28:
Hi!

**Distribution**
NixOS

The output of 
```
lsusb -d 1532: -v
```
is
```
Bus 001 Device 011: ID 1532:0a2e Razer USA, Ltd Razer Wolverine V2 Chroma
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 [unknown]
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a2e Razer Wolverine V2 Chroma
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Wolverine V2 Chroma
  iSerial                 3 000038BE2EBFF800
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0060
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 [unknown]
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

```

The output of the 
```
ls -lah /dev/input/by-id/ | grep Razer
```
is
```
lrwxrwxrwx 1 root root  10 06-28 17:38 usb-Razer_Razer_Wolverine_V2_Chroma_000038BE2EBFF800-event-joystick -> ../event17
lrwxrwxrwx 1 root root   6 06-28 17:38 usb-Razer_Razer_Wolverine_V2_Chroma_000038BE2EBFF800-joystick -> ../js0
```

I can capture the device. What is prefered format of the capture? I could for example provide separate ```.pcapng``` files for each event -- connecting the controller, opening razer app, assigning buttons, assigning colour scheme, etc. Would that be ok?