openrazer/openrazer #1990: Support for Deathstalker V2 PRO TKL

stlhrt, 2022-12-27:
### Device Info

* Name: Deathstalker V2 PRO TKL
* Product Number: RZ03-04370100-R3M1
* Model Number: RZ03-0437
* 
### Device Information
```

Bus 001 Device 028: ID 1532:0298 Razer USA, Ltd DSV2Pro TKL
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0298 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 DSV2Pro TKL
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      75
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx 1 root root  10 Dec 28 00:07 usb-Razer_DSV2Pro_TKL-event-if01 -> ../event26
lrwxrwxrwx 1 root root  10 Dec 28 00:07 usb-Razer_DSV2Pro_TKL-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Dec 28 00:07 usb-Razer_DSV2Pro_TKL-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Dec 28 00:07 usb-Razer_DSV2Pro_TKL-if02-event-mouse -> ../event27
lrwxrwxrwx 1 root root   9 Dec 28 00:07 usb-Razer_DSV2Pro_TKL-if02-mouse -> ../mouse5

```

<details>

```
001:028:003:DESCRIPTOR         1672183378.174976
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0

001:028:002:DESCRIPTOR         1672183378.176019
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 75 08 95 01 81 06 09 30 09 31 16 00 80
 26 FF 7F 75 10 95 02 81 06 C0 C0

001:028:001:DESCRIPTOR         1672183378.177592
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

001:028:000:DESCRIPTOR         1672183378.178794
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]
[image]
[image]

digitwolf, 2022-12-29:
Also `1532:0296` for a different variation of the same keyboard

stlhrt, 2022-12-29:
1532:0296 is for wireless dongle, 0298 is wired ;-)

stlhrt, 2023-01-12:
[dsv2pro_wired_studio_special_keys.pcapng.gz](https://github.com/openrazer/openrazer/files/10397030/dsv2pro_wired_studio_special_keys.pcapng.gz)
[dsv2pro_wired_basics_and_power.pcapng.gz](https://github.com/openrazer/openrazer/files/10397031/dsv2pro_wired_basics_and_power.pcapng.gz)

stlhrt, 2023-01-12:
[dsv2pro_wireless_keys_and_rebinding.pcapng.gz](https://github.com/openrazer/openrazer/files/10397077/dsv2pro_wireless_keys_and_rebinding.pcapng.gz)
[dsv2pro_wireless_chroma_and_power.pcapng.gz](https://github.com/openrazer/openrazer/files/10397078/dsv2pro_wireless_chroma_and_power.pcapng.gz)

digitwolf, 2023-01-19:
@stlhrt is there anything else apart from you pull request that I need to pull locally to test this on my keyboard?

stlhrt, 2023-01-19:
I don't think so...  but it doesn't work very well unfortunately :(

john-2000, 2023-02-08:
what can I do to help ?

meeuw, 2023-02-26:
For what is might help;

report_index / response_index is 0x00 for this device according to the device descriptor (Thanks for providing this!)

meeuw, 2023-02-26:
Could you please provide the output from `usbhid-dump -m 1532 -ed` for both the wireless and wired devices?

meeuw, 2023-02-27:
For what it might help;

report_index / response_index is 0x03 for this device according to the device descriptor (Thanks for providing this!)

john-2000, 2023-02-27:
**Wired:**
_sudo usbhid-dump -m 1532 -ed_

003:011:003:DESCRIPTOR         1677523212.592463
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0

003:011:002:DESCRIPTOR         1677523212.595458
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 75 08 95 01 81 06 09 30 09 31 16 00 80
 26 FF 7F 75 10 95 02 81 06 C0 C0

003:011:001:DESCRIPTOR         1677523212.598456
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

003:011:000:DESCRIPTOR         1677523212.601457
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:008:002:DESCRIPTOR         1677523212.603457
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81
 25 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F
 75 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F
 75 10 95 02 81 06 C0 06 00 FF 09 02 15 00 25 01
 75 08 95 5A B1 01 C0

003:008:001:DESCRIPTOR         1677523212.606460
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 05 15
 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03 06
 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81 25
 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F 75
 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F 75
 10 95 02 81 06 C0 C0 05 0C 09 01 A1 01 85 02 19
 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81 00 75
 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03 19 81
 29 83 15 00 25 01 75 01 95 03 81 02 95 05 81 01
 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85 04 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F 81
 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05 01 09
 00 A1 01 85 08 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85 09
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F
 81 00 C0

003:008:000:DESCRIPTOR         1677523212.609459
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

john-2000, 2023-02-27:
**Wireless 2.4GHz Dongle**
_usbhid-dump -m 1532 -ed_

003:011:003:DESCRIPTOR         1677523212.592463
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0

003:011:002:DESCRIPTOR         1677523212.595458
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 75 08 95 01 81 06 09 30 09 31 16 00 80
 26 FF 7F 75 10 95 02 81 06 C0 C0

003:011:001:DESCRIPTOR         1677523212.598456
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 C0 05 0C 09 01 A1
 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

003:011:000:DESCRIPTOR         1677523212.601457
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

003:008:002:DESCRIPTOR         1677523212.603457
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81
 25 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F
 75 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F
 75 10 95 02 81 06 C0 06 00 FF 09 02 15 00 25 01
 75 08 95 5A B1 01 C0

003:008:001:DESCRIPTOR         1677523212.606460
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 05 15
 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03 06
 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81 25
 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F 75
 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F 75
 10 95 02 81 06 C0 C0 05 0C 09 01 A1 01 85 02 19
 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81 00 75
 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03 19 81
 29 83 15 00 25 01 75 01 95 03 81 02 95 05 81 01
 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85 04 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F 81
 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05 01 09
 00 A1 01 85 08 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85 09
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F
 81 00 C0

003:008:000:DESCRIPTOR         1677523212.609459
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

johnny@johnny-desktop:~$ sudo usbhid-dump -m 1532 -ed
003:008:002:DESCRIPTOR         1677523261.092439
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81
 25 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F
 75 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F
 75 10 95 02 81 06 C0 06 00 FF 09 02 15 00 25 01
 75 08 95 5A B1 01 C0

003:008:001:DESCRIPTOR         1677523261.095438
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 05 15
 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03 06
 00 FF 75 08 95 01 81 03 05 0C 0A 38 02 15 81 25
 7F 75 08 95 01 81 06 05 01 09 38 15 81 25 7F 75
 08 95 01 81 06 09 30 09 31 16 00 80 26 FF 7F 75
 10 95 02 81 06 C0 C0 05 0C 09 01 A1 01 85 02 19
 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81 00 75
 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03 19 81
 29 83 15 00 25 01 75 01 95 03 81 02 95 05 81 01
 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85 04 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F 81
 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05 01 09
 00 A1 01 85 08 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85 09
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F
 81 00 C0

003:008:000:DESCRIPTOR         1677523261.098433
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

dochevviktor, 2023-04-15:
Thanks @aginies for taking the time for the PR with both wired and wireless.
I can confirm it's working on my DeathStalkerV2 Pro TKL after building your version.

PR #2032 (2023-04-24): Add DeathStalkerV2 Pro TKL Wired and Wireless support
Just got one yesterday for testing.
Adding support for Wired and Wireless modes.
Tested successfully with polychromatic

Fixes #1990
closed 2023-04-24 by z3ntu