openrazer/openrazer #1992: Static not working for Razer Blade 15 Advanced (Early 2022)

lah7, 2022-12-28:
Multiple users have reported the hardware static effect does not work for **Razer Blade 15 Advanced (Early 2022)**.

* https://github.com/polychromatic/polychromatic/issues/440
* https://github.com/polychromatic/polychromatic/issues/458

Likely the wrong command is being sent for this particular device model as the LEDs don't change.

z3ntu, 2023-01-09:
@brian-mcnamara Do you also see this on your device?

brian-mcnamara, 2023-01-10:
I'm unable to replicate, was able to select static and select multiple colors. 

@joemckie @Weuxel 
can you send more info about the device you are on and what openrazer is reporting under "Device Info"?

z3ntu, 2023-01-10:
Can you both send your firmware version? Should be part of polychromatic device Info I think

Weuxel, 2023-01-11:
Sure.

[image]

z3ntu, 2023-01-11:
I'd guess for the Blades also Synapse will have the updater built-in? I think I saw someone even upgrading it with USB passthrough to a Windows VM but I can't recommend this since I tried with a Firefly mousepad and that semi-bricked it and I needed a real Windows machine to restore it.

brian-mcnamara, 2023-01-11:
Not that I can tell. I have my blade in a dual boot and synapse says I'm all up to date. Does not look like there is a bios update available too which is my only other idea...

camholl, 2023-01-30:
Hi, I'm having this issue on a 2022 blade 15 base. Switching to another effect and back again works, but switching from static to a different colour does not.
No usb signals are sent to the keyboard during in the latter case, and dbus logs show the following suspicious output:

`method return time=1675081797.008148 sender=:1.91 -> destination=:1.169 serial=719 reply_serial=18
signal time=1675081797.014443 sender=org.freedesktop.DBus -> destination=:1.169 serial=4294967295 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.169"
signal time=1675081797.014467 sender=org.freedesktop.DBus -> destination=(null destination) serial=4294967295 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.169"
   string ":1.169"
   string ""`

Strangely, a similar output appears when switching from "breathing" to static:

`method return time=1675081482.814051 sender=:1.91 -> destination=:1.163 serial=656 reply_serial=19
signal time=1675081482.820733 sender=org.freedesktop.DBus -> destination=:1.163 serial=4294967295 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.163"
signal time=1675081482.820758 sender=org.freedesktop.DBus -> destination=(null destination) serial=4294967295 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.163"
   string ":1.163"
   string ""`

I don't really have the knowledge to know where to begin with this one, but I'll have a look into it. In the meantime, hopefully someone with more knowledge than me can make some sense of this :)

z3ntu, 2023-01-30:
Try `systemctl --user stop openrazer-daemon.service` and then run `openrazer-daemon -Fv`

Then try again what you've done

brian-mcnamara, 2023-03-03:
I just saw Razer [published new firmware and bios](https://mysupport.razer.com/app/answers/detail/a_id/5900). Interestingly enough, this now puts me on firmware version 1.2 so I completely skipped 1.1. Curious if 1.1 had issues, as I am unable to repo this after upgrading.
@Weuxel curious is updating the firmware/bios will solve the issues for you.

Weuxel, 2023-03-03:
@brian-mcnamara It indeed does! Thank you.

Weuxel, 2023-03-03:
As a little side question: Is it normal, that secure boot is enabled again after a bios update? Does Razer reset the bios settings after an update? I had virtualization turned on... that seemes to have survived.

brian-mcnamara, 2023-03-03:
> As a little side question: Is it normal, that secure boot is enabled again after a bios update? Does Razer reset the bios settings after an update? I had virtualization turned on... that seemes to have survived.

Yes, I too experienced this. I had to disable secure boot after upgrading the bios. Glad it seems to have fixed the issue tho!

phealy, 2023-03-08:
FYI, I think the firmware update has been pulled. Not only did it turn Secure Boot back on for me, it turned *off* Intel VMD (which I had configured to put my two drives in a RAID0 so I didn't have to manage the capacity separately)... and then seemingly wrote something to the EFI partition corrupting while the RAID was not configured, corrupting my RAID set.

See [my reddit post](https://www.reddit.com/r/razer/comments/11i66ie/razer_blade_15_2022_firmware_update_202_caution/) for some details.

brian-mcnamara, 2023-03-08:
Yikes, that's not good! Glad you had a backup. I was debating going with hardware raid, but ended up wanting to play around with zfs some more. Curious if the disabling secure boot had something to do with the need to update the EFI partition, but also curious what it changed since I would assume the non-raid layout of the disk would look nothing like a EFI partition.

phealy, 2023-03-08:
Unfortunately it does look like a regular disk at first - the RAID was just doing 256kb chunks, so one of the drives had chunk 0, which contained the GPT and the partition header for EFI right at the beginning.

Interestingly that firmware update left me on 1.1 and the static display doesn't work - I worked around this for now by creating a custom scheme and just setting it static all one color :)

phealy, 2023-04-10:
Just a note that they've [re-published the updates](https://mysupport.razer.com/app/answers/detail/a_id/5900/~/razer-blade-15-%282022%29-%7C-rz09-0421x-support-%26-faqs) and applying them did fix this problem for me. Static now works as expected.

z3ntu, 2023-04-22:
Can we close this issue then?

brian-mcnamara, 2023-04-22:
> Can we close this issue then?

Yea, this issue is resolved with firmware upgrade.

closed 2023-04-22 by z3ntu