openrazer/openrazer #2000: Where are the GPG keys?

Konglomneshued, 2023-01-07:
### Distribution
OpenSUSE Tumbleweed

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I added the repo but unable to verify keys as I can't find them on the site or Github.

z3ntu, 2023-01-07:
The key for Tumbleweed can be found here I think: https://download.opensuse.org/repositories/hardware:/razer/openSUSE_Tumbleweed/repodata/repomd.xml.key

What do you need it for?

Konglomneshued, 2023-01-15:
> The key for Tumbleweed can be found here I think: https://download.opensuse.org/repositories/hardware:/razer/openSUSE_Tumbleweed/repodata/repomd.xml.key
> 
> What do you need it for?

To verify my download.

z3ntu, 2023-01-16:
Is your question answered then?

Konglomneshued, 2023-01-29:
Yes, thank you.

closed 2023-01-29 by z3ntu