openrazer/openrazer #2002: Unable to load the tab properly due to an error

0pLuS0, 2023-01-15:
Hello,

I made a report over at polychromatic, because it was also giving me an issue, but I'm still having an with the mouse.

https://github.com/polychromatic/polychromatic/issues/461

THANKS

z3ntu, 2023-01-15:
> One last problem, everytime I reboot the system, then log back into Slackware, the mouse cursor is stuck, and I constantly have to unplug the USB cable, and plug it into another port to get it to work.

Does this also happen with OpenRazer not installed? It doesn't really sound like something the driver would normally cause.

Tray menu is polychromatic, has little directly to do with this repo

0pLuS0, 2023-01-16:
The cursor will also stick in Windows too, and I have to unlpug it out of the usb port and plug into another one.

I'm also seeing these usb errors using this mouse.

Jan 15 12:49:28 slack kernel: usb 3-1: device descriptor read/64, error -71
Jan 15 12:49:28 slack kernel: usb 3-1: device descriptor read/64, error -71
Jan 15 12:49:28 slack kernel: usb 1-6: config 1 has an invalid interface number: 2 but max is 1
Jan 15 12:49:28 slack kernel: usb 1-6: config 1 has no interface number 1
Jan 15 12:49:28 slack kernel: usb 3-1: device descriptor read/64, error -71
Jan 15 12:49:28 slack kernel: usb 3-1: device descriptor read/64, error -71
Jan 15 12:49:28 slack kernel: usb 3-1: Device not responding to setup address.
Jan 15 12:49:28 slack kernel: usb 3-1: Device not responding to setup address.
Jan 15 12:49:29 slack kernel: usb 3-1: device not accepting address 4, error -71
Jan 15 12:49:29 slack kernel: usb 3-1: Device not responding to setup address.
Jan 15 12:49:29 slack kernel: usb 3-1: Device not responding to setup address.
Jan 15 12:49:29 slack kernel: usb 3-1: device not accepting address 5, error -71
Jan 15 12:49:29 slack kernel: usb usb3-port1: unable to enumerate USB device

This is the first time I've had a mouse cursor stick when logging into the system. I had a first version Baslisk and v2, both never did this. So for now, I'm getting a replacement and getting a new v3 and see how that goes, I'm hoping it's just a bad mouse, as it seems like it.

0pLuS0, 2023-01-16:
BUT, there's still an issue...

I dual boot Slackware 15 & Windows 11 22H2.

If I boot into Windows, then later boot up Slackware, the Keyboard Effects don't work, sometimes the mouse effects get screwed up too.

hmm

z3ntu, 2023-01-16:
Anything about USB read errors are quite certainly not caused by OpenRazer but more either by just the hardware or maybe some other kernel components, but probably mostly hardware.

lah7, 2023-01-16:
Does sound like a hardware problem or bad cable. `-71` is `protocol error`, _whatever that means._

> #define	EPROTO		71	/* Protocol error */
> 
> --- https://github.com/torvalds/linux/blob/master/include/uapi/asm-generic/errno.h#L54

Uninstalling OpenRazer and testing without for a few days would be one way to **fully** rule out the driver being the cause. I can only think that the mouse **could** have been sent a command that worked for a different firmware but causes yours to become "unresponsive" (stuck, mouse crashed ☠️ ) until power cycled. Just speculating though. If it happens even when Windows is booted from power off, then chances are it is hardware.

------

The dual boot scenario (#1807) is just a cat and mouse problem. Synapse on Windows will force the hardware into "software effect", but OpenRazer will switch to "hardware effect". `openrazer-daemon` needs to run (with persistence on) so it can be re-apply previous settings.

If Slackware and the daemon are not playing nice, you could write a script that runs at boot to [echo to the sysfs driver](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#matrix_effect_static---static-effect-mode-write-only) directly maybe.

0pLuS0, 2023-01-16:
I truly belive it's a bad mouse, it locked up also in the BIOS.

I've been having the same issue(s) with the last version of OpenRazer when Dual Booting. Right now the Keyboard Effects don't work, only 'Static'. If I keep refreshing polychromatice and logging in and out of X, sometimes I can get the Keyboard Effects working again, but it's a pain.

Well, for ```persistence on``` this is the razer.conf settings;

[General]
verbose_logging = False

[Startup]
sync_effects_enabled = True
devices_off_on_screensaver = True
battery_notifier = False
battery_notifier_freq = 600
battery_notifier_percent = 33
restore_persistence = True

How do we know if Slackware and the daemon are not playing nice?

I'm not understanding [echo to the sysfs driver](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#matrix_effect_static---static-effect-mode-write-only) what you mean here, and what this does?

THANKS

0pLuS0, 2023-01-16:
When I boot up Slackware, this is how the razer.log looks, are these all the correct log entries?

[razer.log](https://github.com/openrazer/openrazer/files/10429843/razer.log)

When I ran ```openrazer-daemon -rFv``` to restart/refresh, this is how the logs turned out;

[razer.log](https://github.com/openrazer/openrazer/files/10429849/razer.log)

Why does the second log, when I restarted/refreshed look so different?

THANKS

lah7, 2023-01-17:
I don't see anything unusual in the logs, but we're cross-firing two issues here that I'm losing track.

The mouse obviously has dodgy hardware/firmware/connection. Hope this is resolved after a replacement.

For the dual boot, I left a comment in [#1807](https://github.com/openrazer/openrazer/issues/1807#issuecomment-1385877710). I think I see the problem with that scenario now. I left a comment on how to use sysfs which might be worth a try.

---

> How do we know if Slackware and the daemon are not playing nice?

What I really meant was, I don't have experience with Slackware, but I know it doesn't use `systemd` (which most other distros use for making sure the daemon is running as expected). If `openrazer-daemon` is running as the wrong user, or  D-Bus isn't ready, that could be cause of the problems.

I'd try sysfs driver directly for a non-X11 startup, as there's much less to go wrong :)

-----

> If I keep refreshing polychromatic and logging in and out of X, sometimes I can get the Keyboard Effects working again, but it's a pain.

Is it in darkness? Stuck on an effect?  Always possible for the firmware to _act funny_ and reset the brightness to zero, or need setting to a specific effect first, or they want a different colour, before it works again.

Shouldn't be necessary to log out and in, unless you have a stray `openrazer-daemon` running somewhere that's interfering?

0pLuS0, 2023-01-17:
The mouse was defective, so you can close this.

THANKS

closed 2023-01-18 by lah7