openrazer/openrazer #2007: Unable to load module in-spite of secure boot being disabled.

Vixea, 2023-01-19:
### Distribution
EndeavourOS

Latest Aur

### Support

- [ *] Have you tried rebooting?
- [* ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm unable to load module in-spite of not having secure boot enabled.
(Fails with modprobe: ERROR: could not insert 'razerkbd': Operation not permitted)

https://i.imgur.com/oEuWh69.png

z3ntu, 2023-01-21:
Apart from not running `modprobe` as root, I'm not aware of anything that could produce "Operation not permitted" apart from secure boot.

Vixea, 2023-01-22:
I've copy and pasted the command so I don't think that's the issue.

lah7, 2023-01-22:
Is there anything in the kernel log?

    sudo dmesg | grep razer

Is your system a laptop?  Apparently, some BIOSes with an Intel CPU might need "Intel Platform Trust Technology" disabling too, [according to this answer.](https://stackoverflow.com/a/63449219)

Vixea, 2023-01-22:
No, and i don't have an Intel system but an AMD one

z3ntu, 2023-02-11:
@Vixea Have you been able to figure out anything in the last three weeks why it doesn't work? As written before apart from secure boot or not running modprobe as root I don't know of any way it could fail like that.

Vixea, 2023-02-11:
Nope

Vixea, 2023-02-11:
Now I'm getting this 
==> ERROR: Missing 6.1.11-arch1-1 kernel modules tree for module openrazer-driver/3.5.1.

z3ntu, 2023-02-11:
Then you need to install the kernel headers for the kernel you're running. Check that you have `linux-headers` version `6.1.11.arch1-1` installed.

Vixea, 2023-02-11:
I'm running zen, so I think I can just remove linux-headers

Vixea, 2023-02-11:
remind me of the command for modprobe

Vixea, 2023-02-11:
would kernel lockdown make it impossible?

Vixea, 2023-02-11:
zswap.enabled=0 lockdown=integrity threadirqs amdgpu.mcbp=1 amdgpu.ppfeaturemask=0xffffffff

z3ntu, 2023-02-19:
Yes, kernel lockdown is explicitly designed to prevent loading unauthorized kernel modules. That is causing your issues.

closed 2023-02-19 by z3ntu