openrazer/openrazer #2009: Support for Razer Barracuda Pro

cedric-legallo, 2023-01-20:
### Device Information
```

Bus 001 Device 012: ID 1532:053a Razer USA, Ltd Razer Barracuda Pro 2.4
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x053a 
  bcdDevice            0.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00e0
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
[... 200 lines cut ...]
```
Please post the output of
    ls -lah /dev/input/by-id/
```

<details>
```
001:012:003:DESCRIPTOR         1674229348.336620
 06 00 FF 09 00 A1 01 85 01 19 00 29 3F 15 00 26
 FF 00 75 08 95 3F 81 00 19 00 29 3F 15 00 26 FF
 00 75 08 95 3F 91 00 C0 05 0C 09 01 A1 01 85 02
 09 E9 09 EA 09 E2 75 01 95 03 15 00 25 01 81 02
 75 05 95 01 81 03 09 B6 09 CD 09 B5 75 01 95 03
 81 02 75 05 95 01 81 03 09 CF 75 01 95 01 81 02
 75 0F 95 01 81 03 C0

```
</details>

### Packet Captures

that one is complex, i will try to help with it, not sure if i can

### Synapse Screenshots
Sorry don't have a windows env for now, i will try to come back later with the captures

z3ntu, 2023-02-11:
Would be interesting what options Synapse has for this device, doesn't seem to have any RGB which is mostly what other headsets have supported with Synapse.

Luk164, 2023-02-26:
I do not have the PRO variant but I do have the base one so I can post some screenshots. The only thing missing should be the ANC as base barracuda does not have it. Here:

Sound screen:
[image]

Mixer screen:
[image]

Enhancement screen:
[image]

Mic screen:
[image]

Power screen:
[image]

BTW if support for the base barracuda could be implemented in the same move with PRO I would be grateful.

z3ntu, 2023-03-12:
Can you capture USB pcaps when you enable ANC on the device? https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

That way we should be able to see if it's just a firmware via the normal Razer protocol or something else.

caiolcf, 2025-07-10:
Good afternoon everyone!
Two years later. Anyone focused on that? Or the pcaps are still needed?

z3ntu, 2025-07-15:
@caiolcf With pcaps I could suggest which code paths need to be used in OpenRazer but in general device support needs to be added by someone who has the device. https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md also has some content regarding that.

zockerboy1yt1, 2026-02-27:
is anyone still working for the headset? the only thing i need, is the Battery state. 
if this would be possible to add? maybe?

GigaVibeSlopper67, 2026-09-20:
I threw an AI at this for some sessions and got the following values mapped out for the X and the Pro. I just wanted to read out battery, for the Pro i managed to read out and also set Sidetone, ANC and Power Saving. Use with caution (if you want to use any of these findings) as this is 99% AI reverse engineering, the only bad thing that ever happened was wedging the dongle with some commands it assumed to be the channel opener, which I was able to fix by replugging it.

The project is here:
https://github.com/GigaVibeSlopper67/Barracuda_X_Helpers

zockerboy1yt1, 2026-09-20:
Thanks for sharing this!

I used your --json output feature to integrate the Barracuda Pro into [RazerBatteryTray](https://github.com/HoroTW/RazerBatteryTray )  This allows me to use the exact same tray icon design for my headset as I do for my other Razer devices.

For anyone who wants to use this workaround, you just need to replace/add this custom dummy class in your rbt_manager.py file.

Important Note: Make sure to change the self.script_path variable to the exact path where you extracted the[ barracuda_battery.py  ]
(https://github.com/GigaVibeSlopper67/Barracuda_X_Helpers) script on your own machine (replace the /home/zockerboy/... part).

```python
# --- NEW: Custom Class for Barracuda Headset ---

class BarracudaDevice:
    def __init__(self):
        self.name = "Razer Barracuda Pro"
        # CHANGE THIS PATH TO YOUR OWN SYSTEM DIRECTORY:
        self.script_path = "/home/zockerboy/Software/Barracuda_X_Helpers-main/barracuda_battery.py"

    def _get_data(self):
        try:
            cmd = [self.script_path, '--json']
            result = subprocess.run(cmd, capture_output=True, text=True, check=True)
            return json.loads(result.stdout)
        except Exception:
            raise Exception("Barracuda not connected or script failed")

    @property
    def battery_level(self):
        data = self._get_data()
        return data.get("percent", 0)  

    @property
    def is_charging(self):
        data = self._get_data()
        status = data.get("status", "discharging")  
        return status in ["charging", "fully charged"]
# -----------------------------------------------

GigaVibeSlopper67, 2026-09-23:
Glad it it's of use for you! I also most probably found out what the "millivolt value" actually is. It seems to be wireless signal strength, it drops when I step away from the dongle. I've updated the docs accordingly.