openrazer/openrazer #2010: Naga Pro Wireless + Charging dock interaction breaks Naga Pros button panel lighting

BLovegrove, 2023-01-24:
### Distribution

Ubuntu 22.10 (Kinetic Kudu)

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When attaching the Naga Pro Wirelss to the Razer Dock, a breathing effect is added to the rgb lighting on both (official synapse, only mouse for openrazer) device(s) to indicate charging. The color changes between red and green (on a scale through yelolow, orange, etc) to indicate battery level while it charges. It's a neat feature but whenever I remove the mouse from the dock and the lighting goes back to your current preset the side lighting on the 3x4 button panel (labelled 'mouse lighting' or matrix pos 2 in software) defaults to the stock standard razer spectrum effect. 

Can someone either help me work out how to disable the charging effect entirely or diagnose what causes the effect to stick so it can be fixed? 

Thanks in advance! Let me know if theres any extra info you'd like or need.

z3ntu, 2023-02-11:
I'm not aware of any code in OpenRazer that changes the lighting when you put the mouse on the dock. The only thing that might be inconsistent between the different zones is the VARSTORE vs NOSTORE parameter we use when sending an effect to the device, that is normaly used for making the device remember the effect across power cycles vs an ephemeral effect that will be reset after power cycle.

BLovegrove, 2023-02-11:
If it helps, this specific model seems to have no onboard storage for lighting. Im not sure how the code works but I would assume there's some built in behaviour on the mouse that sets the lighting to the charging effect when the dock is detected and openrgb isn't able to set it back automatically, or isn't looking out for that change etc.

closed 2023-07-12 by github-actions[bot]