openrazer/openrazer #2011: razermouse to kernel syslog

sleeplessaa, 2023-01-25:
Hello team,

I found this in my /var/log/syslog : 

Jan 25 16:07:09  kernel: [3237396.953233] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .

Jan 25 16:27:12  kernel: [3238599.049918] razermouse: Command timed out. status: 04 transaction_id.id: ff remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .

The output appear between every 10 - 20 minutes

I can't find any cron task connected.
grep razer /etc/cron* -R 

uname -a
Linux skip 6.1.0 #1 SMP PREEMPT_DYNAMIC Fri Dec 16 22:41:21 EET 2022 x86_64 GNU/Linux

installed packages : 

ii  openrazer-daemon                       3.5.1-1                                    all          OpenRazer peripheral drivers (daemon)
ii  openrazer-driver-dkms                  3.5.1-1                                    all          OpenRazer peripheral drivers (DKMS)
ii  openrazer-meta                         3.5.1-1                                    all          OpenRazer peripheral drivers (metapackage)
ii  python3-openrazer

ii  polychromatic                          0.8.0-2                                    all          RGB lighting utility (metapackage)
ii  polychromatic-cli                      0.8.0-2                                    all          RGB lighting utility (CLI interface)
ii  polychromatic-common                   0.8.0-2                                    all          RGB lighting utility (common files)
ii  polychromatic-controller               0.8.0-2                                    all          RGB lighting utility (Qt interface)
ii  polychromatic-tray-applet              0.8.0-2                                    all          RGB lighting utility (Indicator applet)

hardware : 

Bus 005 Device 007: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
Bus 005 Device 006: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock

Anyone with this issue  ?

z3ntu, 2023-01-25:
We do have some timer in the daemon that checks for battery percentage and sends a notification to the user if it's below the configured threshold. You can adjust this in `~/.config/openrazer/razer.conf`

These kernel messages basically just mean that whatever the kernel driver wanted to send to the device, it didn't get a response. This may or may not be bad depending on how important you think this query was, so in this case the battery % won't be queried successfully. While you can try hacking around the driver by increasing timeouts or something else, generally it's not something to worry about.

Also your mouse might just have gotten to power saving mode where it either takes longer to answer requests or doesn't answer at all anymore.

sleeplessaa, 2023-01-25:
from ~/.config/openrazer/razer.conf

battery_notifier = False
solved the syslog output.

Thank you.

closed 2023-01-27 by z3ntu