openrazer/openrazer #2017: Razer Lancehead Tournament Edition not recognized

VirtualLusamine, 2023-02-07:
### Distribution Manjaro Linux
Openrazer on polychromatic

### Support

- [x] Have you tried rebooting?
I tried rebooting multiple times and it didnt work ever.
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
i tried truble shooting and did all the steps twice or three times and have went through some other support requests.

Im using polychromatic and i have my razer keyboard connected and it works normally, my headset doesnt work but thats for lack of support and it says clearly on the trubleshooting. Although my Lancehead Tournament Edition seems to not apear in the openrazer app trubleshooting as not working but it still says its a unrecognised device. So i really dont know what to do.

[image]
[image]

VirtualLusamine, 2023-02-07:
the update seemed to fix it somehow
really wiered but im just glad it works now

VirtualLusamine, 2023-02-07:
it seems to have come back with rebooting my computer

lah7, 2023-02-08:
`1532:0528` isn't supported by OpenRazer yet, but there is an open issue for it. Please watch #1280 for updates. Apparently this device doesn't have RGB, so it might be limited if it is supported, as OpenRazer doesn't deal with audio-specific features.

Regarding Lancehead TE,  do you mean the problem has come back? Does restarting OpenRazer from that button help?

closed 2023-07-09 by github-actions[bot]