openrazer/openrazer #2019: Lighting effect on logout

alphawave-git, 2023-02-11:
Hi,
I have just set up openrazer and all is working well. I was pleased to find https://github.com/openrazer/openrazer/issues/336 to prevent the lighting switching off at logout
would it be possible to add
1) a custom effect on logout (e.g. a dim version could be used)
2) a 'wake on keypress' which would restore the default effect?

Thanks,

z3ntu, 2023-02-11:
I think you could make a custom e.g. Python script to handle this. But I'm not sure if there's any event you could cling to on logout, also since openrazer daemon should be killed on logout it might be a bit late then.

There's a bunch of Python examples in https://github.com/openrazer/openrazer/tree/master/examples and for reference our screensaver code is here: https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/screensaver_monitor.py which listens to dbus signals being sent from the screensaver.

alphawave-git, 2023-02-12:
Thanks for reply - I was thinking of the same condition that _devices_off_on_screensaver_ manages: so rather than just effect on/off it could have a third 'saver' state

closed 2023-07-13 by github-actions[bot]