openrazer/openrazer #2021: Latest kernel updates seem to have issues with openrazer drivers and polychromatic (possibly –no-initramfs issue) on openSUSE Tumbleweed

iqgrande, 2023-02-17:
Greetings,

Lately the openrazer drivers don't appear to work (e.g., Polychromatic displays "No backends available") on openSUSE Tumbleweed. Since these items aren't automatically rebuilt, I manually uninstall (via `sudo zypper --non-interactive rm $( rpm -qa "openrazer-kernel-modules-dkms*" )`) and then reinstall them (via command below). Anyways, I notice an error/warning message below (observable when it is installed on lastest Tumbleweed) that may be the culprit of this (it looks like something is calling a command with the command line option `--no-initramfs` and the command does not know what that is). Could this be the issue? Thank you for your help with this. The issue can also be observed if you just install them (with the command below). Thank you for your help with this. 

```sh
> sudo zypper in openrazer-meta polychromatic
Loading repository data...
Reading installed packages...
'polychromatic' is already installed.
No update candidate for 'polychromatic-0.8.0-2.2.noarch'. The highest available version is already installed.
Resolving package dependencies...

The following 4 NEW packages are going to be installed:
  openrazer-daemon openrazer-kernel-modules-dkms openrazer-meta python3-openrazer

4 new packages to install.
Overall download size: 257.5 KiB. Already cached: 0 B. After the operation, additional 1.6 MiB will be
used.
Continue? [y/n/v/...? shows all options] (y): y
Retrieving: openrazer-kernel-modules-dkms-3.5.1-2.5.noarch (hardware:razer (openSUSE_Tumbleweed))
                                                                                     (1/4),  60.7 KiB    
Retrieving: openrazer-kernel-modules-dkms-3.5.1-2.5.noarch.rpm .......................[done (60.7 KiB/s)]
Retrieving: openrazer-daemon-3.5.1-2.5.noarch (hardware:razer (openSUSE_Tumbleweed)) (2/4), 145.5 KiB    
Retrieving: openrazer-daemon-3.5.1-2.5.noarch.rpm ...................................[done (112.0 KiB/s)]
Retrieving: python3-openrazer-3.5.1-2.5.noarch (hardware:razer (openSUSE_Tumbleweed))
                                                                                     (3/4),  44.6 KiB    
Retrieving: python3-openrazer-3.5.1-2.5.noarch.rpm ................................................[done]
Retrieving: openrazer-meta-3.5.1-2.5.noarch (hardware:razer (openSUSE_Tumbleweed))   (4/4),   6.7 KiB    
Retrieving: openrazer-meta-3.5.1-2.5.noarch.rpm ...................................................[done]

Checking for file conflicts: ......................................................................[done]
(1/4) Installing: openrazer-kernel-modules-dkms-3.5.1-2.5.noarch ..................................[done]
(2/4) Installing: openrazer-daemon-3.5.1-2.5.noarch ...............................................[done]
(3/4) Installing: python3-openrazer-3.5.1-2.5.noarch ..............................................[done]
(4/4) Installing: openrazer-meta-3.5.1-2.5.noarch .................................................[done]
Creating symlink /var/lib/dkms/openrazer-driver/3.5.1/source -> /usr/src/openrazer-driver-3.5.1

Building module:
cleaning build area...
KERNELDIR=/lib/modules/6.1.10-1-default/build make driver.....
cleaning build area...

razerkbd.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.10-1-default/updates/

razermouse.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.10-1-default/updates/

razerkraken.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.10-1-default/updates/

razeraccessory.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.10-1-default/updates/
Adding any weak-modules
getopt: unrecognized option '--no-initramfs'
Usage: ${0##*/} [options] {--add-modules|--remove-modules}
       ${0##*/} [options] {--add-kernel|--remove-kernel} {kernel-release}

--add-modules
        Add a list of modules read from standard input. Create
        symlinks in compatible kernel's weak-updates/ directory.
        The list of modules is read from standard input.

--remove-modules
        Remove compatibility symlinks from weak-updates/ directories
        for a list of modules.  The list of modules is read from
        standard input.

--add-kernel
        Add compatibility symlinks for all compatible modules to the
        specified or running kernel.

--remove-kernel
        Remove all compatibility symlinks for the specified or current
        kernel.

--verbose
        Print the commands executed.

-dry-run
        Do not create/remove any files.
depmod...
********************************************
* To complete installation, please run:    *
* # sudo gpasswd -a <yourUsername> plugdev *
********************************************

Executing %posttrans scripts ...................................................................<100%>[\i
```

Kind regards,
Anthony

iqgrande, 2023-02-17:
I have some debugging information below. 

```sh
> lsusb | grep 1532
Bus 001 Device 002: ID 1532:0520 Razer USA, Ltd Kraken Tournament Edition
Bus 001 Device 033: ID 1532:005c Razer USA, Ltd DeathAdder Elite

> uname -r
6.1.10-1-default

> sudo dkms status
[sudo] password for root: 
openrazer-driver/3.5.1, 6.1.10-1-default, x86_64: installed

> sudo modprobe razerkbd
> 

> id | grep plugdev | wc -l
1

> lsmod | grep razer
razerkbd              110592  0
razermouse            151552  0
usbcore               380928  11 xhci_hcd,snd_usb_audio,usbhid,snd_usbmidi_lib,razerkbd,usb_storage,uvcvideo,btusb,xhci_pci,uas,razermouse

> dmesg | grep razer
[    5.083328] razermouse 0003:1532:005C.0002: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[    5.279006] razermouse 0003:1532:005C.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[    5.667026] razermouse 0003:1532:005C.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[42193.621299] razermouse 0003:1532:005C.0007: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[42193.680139] razermouse 0003:1532:005C.0008: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[42193.740296] razermouse 0003:1532:005C.0009: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[108410.503752] razermouse 0003:1532:005C.000C: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[108410.562496] razermouse 0003:1532:005C.000D: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[108410.622671] razermouse 0003:1532:005C.000E: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[108774.216088] razermouse 0003:1532:005C.0011: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[108774.275075] razermouse 0003:1532:005C.0012: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[108774.335137] razermouse 0003:1532:005C.0013: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[109189.897344] razermouse 0003:1532:005C.0016: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[109189.956178] razermouse 0003:1532:005C.0017: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[109190.016404] razermouse 0003:1532:005C.0018: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[110087.466981] razermouse 0003:1532:005C.001A: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[110087.526579] razermouse 0003:1532:005C.001B: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[110087.586618] razermouse 0003:1532:005C.001C: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2
[122121.333167] razermouse 0003:1532:005C.001D: input,hidraw1: USB HID v1.11 Mouse [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input0
[122121.392135] razermouse 0003:1532:005C.001E: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input1
[122121.452347] razermouse 0003:1532:005C.001F: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer DeathAdder Elite] on usb-0000:00:14.0-10.2/input2

> sudo journalctl -b | grep -i razer_mount
Feb 15 08:17:56 babel razer_mount[1003]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1004]: Device_ID 1-10.2
Feb 15 08:17:56 babel razer_mount[1005]: Modprobing razermouse
Feb 15 08:17:56 babel razer_mount[1027]: Modprobed razermouse
Feb 15 08:17:56 babel razer_mount[1036]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1037]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1038]: Device_ID 1-10.2:1.0
Feb 15 08:17:56 babel razer_mount[1039]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1041]: Device_ID 1-10.2:1.2
Feb 15 08:17:56 babel razer_mount[1042]: Device_ID 1-10.2:1.1
Feb 15 08:17:56 babel razer_mount[1074]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1073]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1077]: Driver razermouse
Feb 15 08:17:56 babel razer_mount[1091]: Device_ID 0003:1532:005C.0003
Feb 15 08:17:56 babel razer_mount[1090]: Device_ID 0003:1532:005C.0004
Feb 15 08:17:56 babel razer_mount[1093]: Device_ID 0003:1532:005C.0002
Feb 15 08:17:56 babel razer_mount[1097]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
Feb 15 08:17:56 babel razer_mount[1098]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0003/
Feb 15 08:17:56 babel razer_mount[1099]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0002/
Feb 15 08:17:56 babel razer_mount[1111]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0003/
Feb 15 08:17:56 babel razer_mount[1112]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:005C.0004/
[... 198 lines cut ...]
-r--r--r-- 1 root plugdev 4096 Feb 16 18:13 country
[... cut ...]
lrwxrwxrwx 1 root plugdev    0 Feb 16 18:13 driver -> ../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Feb 16 18:13 hidraw
drwxr-xr-x 3 root plugdev    0 Feb 16 18:13 input
-r--r--r-- 1 root plugdev 4096 Feb 16 18:13 modalias
drwxr-xr-x 2 root plugdev    0 Feb 16 18:13 power
-r--r--r-- 1 root plugdev 4096 Feb 16 18:13 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb 16 18:13 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb 16 18:13 uevent

```

iqgrande, 2023-02-17:
The Polychromatic app has a Troubleshooting entry. It mentions the following:

- Failed -> Python library is installed
  - Currently running Python 3.10.9 from /usr/bin/python3

That package is installed (you can see it getting installed from lists above). This may be an issue. What can I check for the health of this?

iqgrande, 2023-02-17:
I uninstalled Polychromatic and installed RazerGenie in its place and things appear to work. If I had to guess, Polychromatic is using the Python bindings (since it is written in PyQt5) whereas RazerGenie is not. This is pointing more towards there being an issue with the `python3-openrazer` package.

z3ntu, 2023-02-19:
Weird, is the python library installed in a wrong directory for an older python version maybe? OBS should normally automatically rebuild the package in that case

iqgrande, 2023-02-19:
Your Python theory sounds possible. If you send me some commands/instruction, then I'll give it a whirl and report back. The only thing I could think to check is what the `python3-openrazer` package installs. 

```sh
> rpm -ql python3-openrazer
/usr/lib/python3.10/site-packages/openrazer
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info/PKG-INFO
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info/SOURCES.txt
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info/dependency_links.txt
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info/requires.txt
/usr/lib/python3.10/site-packages/openrazer-3.5.1-py3.10.egg-info/top_level.txt
/usr/lib/python3.10/site-packages/openrazer/__init__.py
/usr/lib/python3.10/site-packages/openrazer/__pycache__
/usr/lib/python3.10/site-packages/openrazer/__pycache__/__init__.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client
/usr/lib/python3.10/site-packages/openrazer/client/__init__.py
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/__init__.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/constants.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/debug.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/device.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/fx.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/__pycache__/macro.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/constants.py
/usr/lib/python3.10/site-packages/openrazer/client/debug.py
/usr/lib/python3.10/site-packages/openrazer/client/device.py
/usr/lib/python3.10/site-packages/openrazer/client/devices
/usr/lib/python3.10/site-packages/openrazer/client/devices/__init__.py
/usr/lib/python3.10/site-packages/openrazer/client/devices/__pycache__
/usr/lib/python3.10/site-packages/openrazer/client/devices/__pycache__/__init__.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/devices/__pycache__/keyboard.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/devices/__pycache__/mice.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/devices/__pycache__/mousemat.cpython-310.pyc
/usr/lib/python3.10/site-packages/openrazer/client/devices/keyboard.py
/usr/lib/python3.10/site-packages/openrazer/client/devices/mice.py
/usr/lib/python3.10/site-packages/openrazer/client/devices/mousemat.py
/usr/lib/python3.10/site-packages/openrazer/client/fx.py
/usr/lib/python3.10/site-packages/openrazer/client/macro.py

```

lah7, 2023-02-19:
The files for `python3-openrazer` package looks OK. It would be worth checking your Python paths too:

    python3 -c "import sys; print(sys.path)"

> e.g.
>    ['', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/usr/lib/python3.10/site-packages']

Polychromatic's troubleshooter could be reporting the Python library isn't installed if it can't be "imported". Are there any errors if you try manually?

    python3 -c "import openrazer.client"

Such scenario could happen if another library on the system is missing or causing a conflict (like something installed system-wide via `pip`)

iqgrande, 2023-02-19:
Here is the Python path information.

```sh
> python3 -c "import sys; print(sys.path)"
['', '/usr/lib64/python310.zip', '/usr/lib64/python3.10', '/usr/lib64/python3.10/lib-dynload', '/usr/lib64/python3.10/site-packages', '/usr/lib64/python3.10/site-packages/PIL', '/usr/lib64/python3.10/_import_failed', '/usr/lib/python3.10/site-packages']
```

And there are a bunch of errors here. 

```sh
> python3 -c "import sys; print(sys.path)"
['', '/usr/lib64/python310.zip', '/usr/lib64/python3.10', '/usr/lib64/python3.10/lib-dynload', '/usr/lib64/python3.10/site-packages', '/usr/lib64/python3.10/site-packages/PIL', '/usr/lib64/python3.10/_import_failed', '/usr/lib/python3.10/site-packages']
iqgrande@babel:~> python3 -c "import openrazer.client"
Traceback (most recent call last):
  File "/usr/lib64/python3.10/site-packages/numpy/core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/usr/lib64/python3.10/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/usr/lib64/python3.10/site-packages/numpy/core/overrides.py", line 6, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libcblas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/openrazer/client/__init__.py", line 5, in <module>
    from openrazer.client.device import RazerDeviceFactory as _RazerDeviceFactory
  File "/usr/lib/python3.10/site-packages/openrazer/client/device.py", line 4, in <module>
    from openrazer.client.devices import RazerDevice as __RazerDevice, BaseDeviceFactory as __BaseDeviceFactory
  File "/usr/lib/python3.10/site-packages/openrazer/client/devices/__init__.py", line 5, in <module>
    from openrazer.client.fx import RazerFX as _RazerFX
  File "/usr/lib/python3.10/site-packages/openrazer/client/fx.py", line 3, in <module>
    import numpy as _np
  File "/usr/lib64/python3.10/site-packages/numpy/__init__.py", line 141, in <module>
    from . import core
  File "/usr/lib64/python3.10/site-packages/numpy/core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.10 from "/usr/bin/python3"
  * The NumPy version is: "1.24.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libcblas.so.3: cannot open shared object file: No such file or directory
```

It looks like NumPy is installed. 

```sh
> rpm -qa | grep -i numpy
python38-numpy-1.24.1-3.1.x86_64
python39-numpy-1.24.1-3.1.x86_64
python310-numpy-1.24.1-3.1.x86_64
```

It also looks like BLAS is installed. 

```sh
> rpm -qa | grep -i blas
libcblas3-3.9.0-6.1.x86_64
libgslcblas0-2.7.1-1.1.x86_64
libblas3-3.9.0-6.1.x86_64
libopenblas_pthreads0-0.3.21-3.1.x86_64
```

Does this mean that the `python310-numpy` is broken (i.e., it is built against a BLAS that is not present)?

lah7, 2023-02-19:
Looks like  `libcblas3` recently broke for existing Tumbleweed installations. I found this bug report which mentions a command to relink the missing libraries:

* https://bugzilla.opensuse.org/show_bug.cgi?id=1208192
* https://www.reddit.com/r/openSUSE/comments/1117q2w/tw_numpy_complains_about_libcblasso3_cannot_open/

iqgrande, 2023-02-20:
The commands seemed to work!

z3ntu, 2023-03-12:
Can we close this then?

iqgrande, 2023-03-12:
Yes. Thank you for your help.Kind regards,Anthony(Please excuse typos since this was sent from a portable device.)

closed 2023-03-12 by z3ntu