openrazer/openrazer #2022: Support for Razor Blackshark V2 Pro

Nezmin2, 2023-02-17:
### Device Info

* Name: Razor Blackshark V2 Pro
* Product Number: Unknown
* Model Number: RZ04-03220100-R3U1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 003: ID 1532:0528 Razer USA, Ltd Razer BlackShark V2 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0528 
  bcdDevice            1.00
  iManufacturer           3 Razer
  iProduct                4 Razer BlackShark V2 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0132
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0047
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
[... 256 lines cut ...]
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```

lrwxrwxrwx 1 root root   9 Feb 17 08:36 usb-Razer_Razer_BlackShark_V2_Pro-event-if03 -> ../event8

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]
[image]

z3ntu, 2023-03-12:
Duplicate of #1280

closed 2023-03-12 by z3ntu

marcosilva85, 2024-01-24:
@Nezmin2
I assume you have the blackshark v2 pro 2023 version, which linux distro do you use?
 
- how well does this model worked for you in linux, is it good in music and films?
- does the mic performs in [high-quality like in this video](https://youtu.be/7_xjpBYwm5g?t=307)?