openrazer/openrazer #2024: Support for Blackwidow V3 Pro wireless(PID1532:025c)  on Ubuntu 20.04

shhnwz, 2023-02-20:
### Device Info

* Name: Blackwidow v3 Pro wireless keyboard
* Product Number: RZ03-0353
* Model Number:  RZ03-03530100

Polychromatic is not able to detect this device.

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
shhnwz@p3571:~ $ lsusb -d 1532: -v

Bus 003 Device 007: ID 1532:007d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x007d 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V2 Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 385 lines cut ...]
 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85 04 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F 81
 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05 01 09
 00 A1 01 85 08 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85 09
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 0F
 81 00 C0

003:006:000:DESCRIPTOR         1676911492.032309
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

shhnwz, 2023-02-20:
[image]

nightsky30, 2023-02-21:
Duplicate of #1279 and potentially fixed with #1901

z3ntu, 2023-03-12:
Duplicate of #1279

closed 2023-03-12 by z3ntu