openrazer/openrazer #2026: Ghosted keys on BlackWidow Elite when holding down button on Naga Trinity

indie2718, 2023-02-24:
### Distribution
Ubuntu 20.04
OpenRazer 3.5.1

### Support
I've tried going through the troubleshooting steps at the wiki page to no avail, as have I tried rebooting a few times to get changes to take.

I'm not sure what info to provide to show that this is happening, or why it's happening, but when I hold down a mouse button on my Naga Trinity (1532:0067), about 40% of my inputs get ghosted by my BlackWidow Elite (1532:0228). It looks like support has been added for both devices, so I'm not sure what's going on. I'm happy to provide whatever's needed to diagnose.
Thanks!!

z3ntu, 2023-03-12:
Can you try uninstalling OpenRazer and see if the same issue persists?

indie2718, 2023-03-13:
no dice. the problem persists regardless.

z3ntu, 2023-04-22:
Yeah then it very much sounds like a hardware or firmware issue. Nothing we can fix, sorry.

closed 2023-04-22 by z3ntu