openrazer/openrazer #2027: pylib: scroll_wheel property should be named scroll

henryruhs, 2023-02-24:
This inconsistency forces me to process a dict to process through possible misc effects.

```
@property
def scroll_wheel(self):
    return self._scroll
```

https://github.com/openrazer/openrazer/blob/master/pylib/openrazer/client/fx.py#L1028-L1030

How does my code look?

```
def set_device(device : Any, color_config : Color) -> bool:
	if device.has('brightness'):
		device.brightness = color_config['brightness'][0]
	if device.fx.has('static'):
		return device.fx.static(color_config['rgb'][0], color_config['rgb'][1], color_config['rgb'][2])
	return use_effect(device, color_config)
```

```
def use_effect(device : Any, color_config : Color) -> bool:
	parts =\
	{
		'logo': 'logo',
		'scroll': 'scroll_wheel',
		'left': 'left',
		'right': 'right',
		'charging': 'charging',
		'fast_charging': 'fast_charging',
		'fully_charged': 'fully_charged',
		'backlight': 'backlight'
	}

	for part_key, part_value in parts.items():
		if device.fx.has(part_key + '_static'):
			effect_function = getattr(getattr(device.fx.misc, part_value), 'static')
			return effect_function(color_config['rgb'][0], color_config['rgb'][1], color_config['rgb'][2])
	return False
```

PS: I wonder if there is a smarter way to set all devices to a static color?

z3ntu, 2023-03-12:
I have a long-term plan of reworking the Python API (and the DBus API also for that matter) to have something more like `zones[0]` instead of `logo` and `zones[1]` instead of `scroll_wheel` or something. Unfortunately too many other things to do so I haven't gotten to that yet.

For now I think you can sort of monkey patch this in in your script, something like this
```python
>>> b
<RazerMouse PM1908H10005919>
>>> b.fx.misc.scroll_wheel.effect
'spectrum'
>>> b.fx.misc.scroll = b.fx.misc.scroll_wheel
>>> b.fx.misc.scroll.effect
'spectrum'
```

But if you send a PR adding the alias in the pylib I'd probably accept it.

z3ntu, 2023-07-21:
Could you send a patch adding this? See my last comment.

closed 2025-01-22 by henryruhs