openrazer/openrazer #2031: Support remapping/binding of keys

meeuw, 2023-02-28:
I found that new Razer Keyboards support remapping/binding of keys.

For example you can swap the Y and Z buttons or make the X button send a mouse click.

This is stored in the device memory and also works in USB boot mode (ie. in your BIOS setup / game console etc)

There seem to be two different commands for analog and non analog keyboards.

Command Class 0x02 / Command ID 0x12 is for analog keyboards
Command Class 0x02 / Command ID 0x0d is for non analog keyboards.

After this the following values are used:

* on-board memory slot (white/active = 1, red = 2, green = 3, blue = 4, cyan = 5)
* Razer key identifier
* Fn / Hypershift flag
* Actuation point
* Release point
* Type (0 = disable, 1=mouse, 2=keyboard, 10=multimedia, 11=doubleclick, 12=fn)
* Parameter length

After this the specific parameters depending on the type must be supplied, this will contain what event should be send.

Would this something be that could be implemented in OpenRazer or is it out of scope for this project?

How should this be implemented? I'm not sure how this should be neatly done through sysfs.

BleedTheFreak, 2023-03-21:
i was marked the caps_lock as fn button but it did not work

meeuw, 2023-03-24:
@BleedTheFreak how did you try this?

I've remapped LEFT_META to Fn and that works, note that you need to use type=12.

For this I'm using the following packets:
```
\x02\x12\x01\x7f\x00\x00\x00\x0c\x01\x01\x00\x00\x00\x00
\x02\x12\x01\x7f\x01\x00\x00\x0c\x01\x01\x00\x00\x00\x00
```

`\x7f` is LEFT_META, for CAPS_LOCK you should use `\x1e` instead.

meeuw, 2023-03-24:
@z3ntu I'm thinking of implementing an interface similar to the Roccat HID driver:

> Module roccat is a char device used to report special events of roccat hardware to userland. These events include requests for on-screen-display of profile or dpi settings or requests for execution of macro sequences that are not stored in device. The information in these events depends on hid device implementation and contains data that is not available in a single hid event or else hidraw could have been used. 
> **It is inspired by hidraw, but uses only one circular buffer for all readers.**

https://github.com/torvalds/linux/blob/cb7f5b41f8341148050fe63e27cf52aa4f1519ad/drivers/hid/hid-roccat.c#L18

Agreeable-f, 2023-04-14:
piper does a (somewhat) good job of this and even has profiles. switching from a logitech g502 to a razer basilisk v3 lost that functionality as libratbag doesn't support razer mice. would be nice to see it for razer mice using openrazer

z3ntu, 2023-04-22:
@meeuw I don't really have any preference there, but if there's already something similar that would be nice if it was at least similar to that.

Mte90, 2023-05-07:
There are cases like with the tartarus where you have the scroll wheell and you want to remap them to keyboard buttons too.

z3ntu, 2023-05-31:
Relevant for here: https://github.com/openrazer/openrazer/pull/2054#issuecomment-1568951655

Mte90, 2023-09-29:
the ticket is still valid bot...

dm17, 2024-01-28:
I don't understand the point of listing mice in openrazer. What exactly is the difference between using a razer mouse with openrazer or the default linux drivers? The tool `evhz` shows the polling rate of a mouse, and it can achieve 1000hz with or without openrazer. And it seems openrazer cannot change the key bindings/mappings of the mice. So I'm honestly curious: what is the supposed purpose of using openrazer on Linux for Razer mice?

Mte90, 2024-01-29:
Because you want to remap buttons everywhere, if the project scope is to replicate the original tool, this is something that does.

jcolivo, 2024-08-24:
> So I'm honestly curious: what is the supposed purpose of using openrazer on Linux for Razer mice?

RGB support for mice that have them.

dm17, 2024-08-24:
I don't want lights on my computer or peripherals... But I do care about the functionality, like the pointer drivers taking full advantage of the hardware and reassigning the mouse buttons.

sealvessel, 2024-10-02:
> I don't want lights on my computer or peripherals... But I do care about the functionality, like the pointer drivers taking full advantage of the hardware and reassigning the mouse buttons (which I didn't see in openrazer for my device). I would like to know if the default Linux driver that is used for Razer mice is really taking full advantage of the hardware like the full featured Windows driver that was used to test Razer hardware in basically all reviews.

I was looking for this too, appears there is no option.

geezmolycos, 2024-10-24:
I have reverse-engineered all commands of Basilisk V3 mouse for button binding. Most recent mice of different models should have similar protocols for binding buttons. It's free for someone more experienced to implement the protocol. The document is in my web app project for changing settings of Basilisk V3 mouse: <https://github.com/geezmolycos/razerqdhid/blob/main/docs/cmd_button-en.md>. It is AI translated from Chinese. If there are any mistakes, you can make an issue.

The original issue is about keyboards and I don't have experience with Razer keyboards. However I think that this can be inspirational for someone to figure out how keyboard remapping works.

rom4ster, 2025-01-02:
> > So I'm honestly curious: what is the supposed purpose of using openrazer on Linux for Razer mice?
> 
> RGB support for mice that have them, and potentially remapping buttons.

If RGB support is all you need, simply use OpenRGB lol. I think that not having button remapping puts this software in a  :( spot. I would like to advocate for button remapping support as most razer mice have at least 2 extra buttons if not more.

theyear2525, 2025-04-15:
> If RGB support is all you need, simply use OpenRGB lol. I think that not having button remapping puts this software in a :( spot. I would like to advocate for button remapping support as most razer [...]

Yes, I must say I'm a bit bummed, too. "exposes all firmware capabilities"...doesn't seem so.
Input Remapper can't remap event's that don't get recognized. Judging also by my previous mouse, some buttons don't get recognized unless you've already bound a key to them using the original manufacturer's software tool.
I bought a Basilisk V3 because I thought I could program mouse buttons with OpenRazer without a VM. Turns out some of the buttons don't even get recognized by evtest (see above) and certain RGB settings make the mouse buggy so it doesn't get recognized on boot and even freezes the PC during wake from suspend (on certain usb ports).
This mouse shouldn't be in the stable list.
Off to searching my win10 key to set  up a VM -.- .

rom4ster, 2025-04-15:
> > > > So I'm honestly curious: what is the supposed purpose of using openrazer on Linux for Razer mice?
> > > 
> > > 
> > > RGB support for mice that have them, and potentially remapping buttons.
> > 
> > 
> > If RGB support is all you need, simply use OpenRGB lol. I think that not having button remapping puts this software in a :( spot. I would like to advocate for button remapping support as most razer mice have at least 2 extra buttons if not more.
> 
> Yes, I must say I'm a bit bummed, too. "exposes all firmware capabilities"...doesn't seem so. Input Remapper can't remap event's that don't get recognized. And judging also by my previous mouse, some buttons don't get recognized unless you've already bound a key to them using the original manufacturer's software tool. I bought a Basilisk V3 because I thought I could program mouse buttons with OpenRazer without a VM. Turns out some of the buttons don't even get recognized by evtest (see above) ~and certain RGB settings make the mouse buggy so it doesn't get recognized on boot and even freezes the PC during wake from suspend (on certain usb ports). This mouse shouldn't be in the stable list.~ [<- I have been notified that this has been fixed with OpenRazer version 3.8.0 - later than I thought, sorry for that] Off to searching my win10 key to set up a VM -.- .
> 
> [I'm sorry I'm bummed, the burden to make mice work in Linux should be on the original manufacturer, not enthusiasts who might expose themselves to some ire in trying to be helpful. But also: the firmware capabilities thing might oversell this driver a bit]

I disagree, I see no reason for razer to make something "work" on linux. Rather, razer should PUBLISH the documentation needed to send custom commands to this mouse. That is INFINITELY more valuable than "oh port your software". To kmy knowledge for example, no one knows how the magneto scroll wheel thing works for razer. Even if you know how the original sensor works, razer does something weird and ive found no one who knows. It is frustrating.

rom4ster, 2025-04-16:
> I didn't rule that out, maybe "make mice work" was a poor choice of words there. What I wanted to say is that I didn't want to poop on the OpenRazer devs because I was acting bummed by expecting it to also reprogram buttons. Of course documentation is always good. As a proponent of right to repair I'm all for it. But for me right now I'd be thrilled even with just a ported software. Or any kind of mouse manufacturers that has full functionality in Linux without shenanigans like this. Funny/stupid how gaming isn't a problem anymore, but mice are ^^; . (at this point I could just ask: are there any manufacturers who support Linux at all? I've heard of Roccat, but they ended support when they got swallowed I think.)
> 
> I wouldn't be able to write a driver from their documentation. geezmolycos above seemed to have found out the necessary commands for the Basilisk, but I can't even begin to understand how to use his findings.
> 
> I had no idea non-standard mouse buttons were such a pita even to be recognized on a low level. Some of my buttons in are related to mouse functionality (specifically: hyper-shift, wheel mode, dpi switch, profile switch). I suspect they communicate with the mouse directly and are not even giving any output to the computer in this default state. They need to be configured on a firmware level first.
> 
> My previous mouse was a Sharkoon SGM1. After switching to Linux at a point, I noticed that Input Remapper wouldn't pick up the additional buttons. I made a VM and set the buttons to keypad numbers. Just confirmed that these get picked up by evtest and Input Remapper. One of the buttons I set to dpi cycle. This one indeed doesn't get picked up. In contrast to the Basilisk mouse, on which I didn't reprogram the buttons with the original software, this one button I deliberately set to that "mouse internal" function. So yea, seems indeed like that's the reason certain buttons are out of reach like that :( .
> 
> I've still no idea where my win10 key is :/ .

Mouse buttons can be very easy or very annoying. Razer for some reason felt the need to checksum their commands for example, which is honestly just extra. On the other hand, this software's value proposition is very low. It only gives you such a small slice of the mouse capabbilities. I wont poop on the devs too much because I dont use linux but I think people like you who are afffected should. Right now the biggest thing this software does is RGB, which is really worthless considering openrgb exists. I think that if buttons are not considered in scope after 8+ years, people should start speaking up and voicing there frustrations. 

I will say geezmolycos was SUCH A HELPFUL DUDE. His repo allowed me to reverse engineer a lot of my mouse. I could have done almost all of it but I was lazy lol. ofc the magneto stufff is a mystery but geez is a legend.  I made a bbunch of diagrams on my repo for my mouse which hopefully will help people with this project.

As for a driver... it might bbbe necessary bbbut you could use non driver code to do most stufff.  IDK if openrazer implements lamparray but if you are on linux, do you really even care? Other than that, idk what a driver can do that user software cant. maybe perfofrmance bonuses.

cD1rtX3, 2025-07-04:
+Infinity. I used Synapse to do a lot of things, including Hypershift shenanegans with my keyboard, and I am disappointed (with Razer) that I can't do that on Linux.