openrazer/openrazer #2033: Battery notifications seem broken

thomasfinstad, 2023-03-04:
### Distribution

Ubuntu 20.04
Python 3.8.10

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have only ever seen one notification for low battery, that was after a reboot.  The mouse was 97% charged but the notification said 0%.

I also get these logs about the notification lib:

```log
$ journalctl -b0 -t openrazer-daemon 
-- Logs begin at Sun 2022-12-04 12:35:11 CET, end at Sat 2023-03-04 07:41:56 CET. --
Mar 04 07:34:10 dsk01 openrazer-daemon[9664]: Starting daemon.
Mar 04 07:34:10 dsk01 openrazer-daemon[9664]: 2023-03-04 07:34:10 | razer                          | INFO     | Initialising Daemon (v3.5.1). Pid: 9664
Mar 04 07:34:10 dsk01 openrazer-daemon[9664]: 2023-03-04 07:34:10 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Mar 04 07:34:10 dsk01 openrazer-daemon[9664]: 2023-03-04 07:34:10 | razer                          | INFO     | Found device.0: 0003:1532:00B7.0006
Mar 04 07:34:10 dsk01 openrazer-daemon[9664]: 2023-03-04 07:34:10 | razer.device0                  | INFO     | Initialising device.0 RazerDeathAdderV3ProWireless
Mar 04 07:34:12 dsk01 openrazer-daemon[9664]: 2023-03-04 07:34:12 | razer                          | INFO     | Serving DBus
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]: Traceback (most recent call last):
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]:     self._handler(*args, **kwargs)
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/notify2.py", line 154, in _closed_callback
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]:     n = notifications_registry[nid]
Mar 04 07:34:23 dsk01 openrazer-daemon[9664]: KeyError: 4
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]: Traceback (most recent call last):
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]:     self._handler(*args, **kwargs)
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/notify2.py", line 154, in _closed_callback
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]:     n = notifications_registry[nid]
Mar 04 07:34:28 dsk01 openrazer-daemon[9664]: KeyError: 1
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]: Traceback (most recent call last):
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]:     self._handler(*args, **kwargs)
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/notify2.py", line 154, in _closed_callback
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]:     n = notifications_registry[nid]
Mar 04 07:34:34 dsk01 openrazer-daemon[9664]: KeyError: 6
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]: Traceback (most recent call last):
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]:     self._handler(*args, **kwargs)
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/notify2.py", line 154, in _closed_callback
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]:     n = notifications_registry[nid]
Mar 04 07:37:13 dsk01 openrazer-daemon[9664]: KeyError: 3
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]: Traceback (most recent call last):
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]:     self._handler(*args, **kwargs)
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]:   File "/usr/lib/python3/dist-packages/notify2.py", line 154, in _closed_callback
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]:     n = notifications_registry[nid]
Mar 04 07:37:17 dsk01 openrazer-daemon[9664]: KeyError: 5
```

**Other logs mentioned in troubleshooting section:**

kernel modules

```
$ lsmod | grep razer
razermouse            122880  0
hid                   147456  5 usbhid,hid_generic,nzxt_kraken2,razermouse
```

[... 94 lines cut ...]
lrwxrwxrwx 1 root plugdev    0 Mar  4 07:33 driver -> ../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Mar  4 07:33 hidraw
drwxr-xr-x 3 root plugdev    0 Mar  4 07:33 input
-r--r--r-- 1 root plugdev 4096 Mar  4 07:33 modalias
drwxr-xr-x 2 root plugdev    0 Mar  4 07:33 power
-r--r--r-- 1 root plugdev 4096 Mar  4 07:33 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Mar  4 07:33 subsystem -> ../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Mar  4 07:33 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:00B7.0008/':
total 0
drwxr-xr-x 5 root plugdev    0 Mar  4 07:33 .
drwxr-xr-x 5 root root       0 Mar  4 07:33 ..
-r--r--r-- 1 root plugdev 4096 Mar  4 07:33 country
lrwxrwxrwx 1 root plugdev    0 Mar  4 07:33 driver -> ../../../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Mar  4 07:33 hidraw
drwxr-xr-x 3 root plugdev    0 Mar  4 07:33 input
-r--r--r-- 1 root plugdev 4096 Mar  4 07:33 modalias
drwxr-xr-x 2 root plugdev    0 Mar  4 07:33 power
-r--r--r-- 1 root plugdev 4096 Mar  4 07:33 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Mar  4 07:33 subsystem -> ../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Mar  4 07:33 uevent
```

I am part of the `plugdev` group as well.

thomasfinstad, 2023-03-12:
> When this happens is the mouse in the normal active mode, not unused and has suspended itself?

I don't really know, is there a log entry I can look up?

z3ntu, 2023-03-12:
Maybe in other words, do you get the 0% notification while actively using the mouse, or only when you're not using the mouse (i.e. it went into power saving mode)?

thomasfinstad, 2023-03-14:
I get no notification at all, no matter the level or state.

HoroTW, 2023-03-14:
@thomasfinstad Can you share your `~/.config/openrazer/razer.conf`

thomasfinstad, 2023-03-14:
@HoroTW  not too much there, anything I should change to help out?

```sh
$ cat ~/.config/openrazer/razer.conf
[General]
verbose_logging = False

[Startup]
sync_effects_enabled = True
devices_off_on_screensaver = True
battery_notifier = True
battery_notifier_freq = 600
battery_notifier_percent = 35
restore_persistence = True
```

HoroTW, 2023-03-14:
Just to ensure that the problem really exists you could change the freq and percent
```
battery_notifier_freq = 60
battery_notifier_percent = 100
```
And restart the daemon by:
```
systemctl --user restart openrazer-daemon.service
```
Then you should (if it works) receive a notification every 60 seconds.
If not please attach the log from (done after the restart of the daemon) `journalctl --user-unit=openrazer-daemon.service --no-pager --since=-2h`

thomasfinstad, 2023-03-15:
I see the notification now so I don't know if maybe I just missed it before, probably did as I can see there is no notification in the "history" or notifier widget (kde plasma), but I did see the pop-up.

I assume this means the issue can be closed, and I will just keep the notifier freq at 60 and set the level back to 35 in the hopes I will catch it next time by having it show up more frequently.

HoroTW, 2023-03-15:
I'm glad it's fixed!
Yeah, if I remember correctly, in KDE only one of the notification centers has the history (e.g. if you have an extra dock or multiple monitors with multiple taskbars (and notification icons), only one of the notification centers has the notification history - or something like that. ^^

I'm also not the biggest fan of notifications for monitoring my mouse battery, so I wrote https://github.com/HoroTW/RazerBatteryTray to show the battery in the tray.

thomasfinstad, 2023-03-15:
@HoroTW I do have multiple notification widgets (one in the dock and one as a "window" on the desktop, but non of them seem to have shown the notification, only the popup.
I do agree however, a systray with the status is far better, thanks I will use that!

closed 2023-03-15 by thomasfinstad

lah7, 2023-03-15:
The missing notification reminds me of an older issue (#1796). There might be a KDE bug whereby if the notification times out, no more would appear until `openrazer-daemon` is relaunched. This wouldn't happen if the notification was explicitly dismissed ❌

Integrating with the desktop's "battery" indicator would be even better: 
* #87