openrazer/openrazer #2034: Could you please add function of mouse macro?

sajia23, 2023-03-04:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

meeuw, 2023-03-11:
Have you tried configuring mouse macros on a Windows machine using Synapse for your mouse? 

I know at least some devices are able to execute macros from hardware.

If you'd like to be able to configure/change macros from Linux I'd recommend to add that to this issue description.

SuperJugy, 2023-05-13:
I don't know about OP, but I would like being able to configure/change macros from Linux for my Razer Viper mouse. I don't think my mouse has hardware support for macros, plus I don't have a Windows machine to do it.

Thanks.

closed 2023-10-11 by github-actions[bot]