openrazer/openrazer #2039: OpenRazer do not run on NixOS Linux

mbledkowski, 2023-03-22:
```console
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/etc/systemd/user/openrazer-daemon.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-03-22 17:42:47 CET; 1min 4s ago
       Docs: man:openrazer-daemon(8)
    Process: 656157 ExecStart=/nix/store/rbgain0755p3svx79xmz4r4ikvlydsh2-python3.10-openrazer_daemon-3.5.1/bin/openrazer-daemon --config /nix/store/nhnqhg186p0ii7bmnz2gkq1v12c38135-razer.conf --foreground (code=exited, status=1/FAILURE)
   Main PID: 656157 (code=exited, status=1/FAILURE)
        CPU: 210ms

Mar 22 17:42:47 jupiter systemd[1848]: openrazer-daemon.service: Scheduled restart job, restart counter is at 5.
Mar 22 17:42:47 jupiter systemd[1848]: Stopped Daemon to manage razer devices in userspace.
Mar 22 17:42:47 jupiter systemd[1848]: openrazer-daemon.service: Start request repeated too quickly.
Mar 22 17:42:47 jupiter systemd[1848]: openrazer-daemon.service: Failed with result 'exit-code'.
Mar 22 17:42:47 jupiter systemd[1848]: Failed to start Daemon to manage razer devices in userspace.
```

z3ntu, 2023-04-01:
Take a look at the full systemd log for this, there should be an error message somewhere. Or run the ExecStart command manually.

closed 2023-08-31 by github-actions[bot]

mbledkowski, 2023-08-31:
D: