openrazer/openrazer #2041: Support for Razer Naga Pro v1.1

Nezmin2, 2023-03-28:
### Device Info

* Name: Naga Pro Wired
* Product Number: RZ01-03420100
* Model Number: RZ01-0342
* OS: PopOS

Scroll wheel behaving erratically. Whenever I push up it goes down then up then down again by itself. It does the same when I push it down. There's no rhyme nor reason to it.

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 002 Device 004: ID 1532:008f Razer USA, Ltd Razer Naga Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x008f 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                2 Razer Naga Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 789 lines cut ...]
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:004:001:DESCRIPTOR         1680023015.746203
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:004:000:DESCRIPTOR         1680023015.749203
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

z3ntu, 2023-04-22:
Sorry, I don't quite understand what your issue is. The device is already supported since #1692 so if you're running a recent-ish version the device is supported.

If you're just talking about this

> Scroll wheel behaving erratically. Whenever I push up it goes down then up then down again by itself. It does the same when I push it down. There's no rhyme nor reason to it.

this just sounds like a hardware issue, what does OpenRazer have to do with it?

z3ntu, 2023-04-28:
As an easy test you can uninstall openrazer from your machine, reboot and see if the problem persists.  
If it does it's either caused by some other software (kernel, libinput, etc) or a hardware issue. Can you please try that?

Nezmin2, 2023-05-01:
> As an easy test you can uninstall openrazer from your machine, reboot and see if the problem persists. If it does it's either caused by some other software (kernel, libinput, etc) or a hardware issue. Can you please try that?

Ok, so I uninstalled openrazer and tried my scroll-wheel without it. The same thing happened. Thanks for helping me troubleshoot the issue at least. I'm a little frustrated at the fact that this mouse isn't that old and it's already borked.

closed 2023-05-01 by Nezmin2

lah7, 2023-05-03:
My mouse (Mamba TE) is old and has had the erratic scrolling problem. Taking it apart to clean for dust and resolder a loose connection to the scroll sensor has helped.