openrazer/openrazer #2043: Razer Naga Pro (Wireless) not found

c0d3m0nky, 2023-04-04:
### Distribution

    Name: Razer Naga Pro (Wireless)
    Product Number: RZ01-03420100
    Model Number: RZ01-0342
    USB VID:PID: 1532:0090
    Version: 3.2.0

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My Naga isn't seen by RazerGenie, razer-cli, don't see anything about it in the logs, nor do I see errors in the logs. I've gone through the troubleshooting steps with nothing wrong as far as I can tell, and it does come up in the list from lsusb with pid 1532:0090 which is in the list of stable devices. Also, I do see the dock and can edit it's settings with no issues, so it seems to me that it's working, just not for the mouse

Below are results from some of the troubleshooting steps I think may be pertinent (left out the greps for brevity)

`lsusb`

Bus 002 Device 052: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 002 Device 049: ID 1532:0090 Razer USA, Ltd Razer Naga Pro

`dkms status`

openrazer-driver/3.2.0, 5.17.15-76051715-generic, x86_64: installed
openrazer-driver/3.2.0, 5.17.5-76051705-generic, x86_64: installed
openrazer-driver/3.2.0, 5.18.10-76051810-generic, x86_64: installed
openrazer-driver/3.2.0, 5.19.0-76051900-generic, x86_64: installed
openrazer-driver/3.2.0, 6.0.12-76060006-generic, x86_64: installed
openrazer-driver/3.2.0, 6.0.6-76060006-generic, x86_64: installed
openrazer-driver/3.2.0, 6.2.6-76060206-generic, x86_64: installed

`lsmod`

razerkbd              102400  0
razeraccessory         73728  0
hid                   176128  4 usbhid,razerkbd,hid_generic,razeraccessory

`dmesg` (this one requires me to sudo, the guide doesn't specify to use sudo. Maybe this is something?)

without sudo

dmesg: read kernel buffer failed: Operation not permitted

with sudo

[    8.560314] razeraccessory 0003:1532:007E.0008: input,hidraw7: USB HID v1.11 Mouse [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input0
[    8.772162] razeraccessory 0003:1532:007E.0009: input,hidraw8: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input1
[    9.012309] razeraccessory 0003:1532:007E.000A: input,hidraw9: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input2
[   72.956289] razeraccessory 0003:1532:007E.0015: input,hidraw12: USB HID v1.11 Mouse [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input0
[   73.020086] razeraccessory 0003:1532:007E.0016: input,hidraw13: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input1
[   73.080277] razeraccessory 0003:1532:007E.0017: input,hidraw14: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:0b:00.0-1.4.2/input2

`cat ~/.local/share/openrazer/logs/razer.log`

2023-04-04 15:05:09 | razer                          | INFO           | Initialising Daemon (v3.2.0). Pid: 2788
2023-04-04 15:05:09 | razer.screensaver    | INFO           | Initialising DBus Screensaver Monitor
2023-04-04 15:05:09 | razer                          | INFO           | Found device.0: 0003:1532:007E.0008
2023-04-04 15:05:09 | razer.device0            | INFO           | Initialising device.0 RazerMouseDock
2023-04-04 15:05:09 | razer                          | INFO           | Serving DBus
2023-04-04 15:05:11 | razer                          | WARNING  | Removing 0003:1532:007E.0008
2023-04-04 15:05:43 | razer                          | INFO           | Found valid device.0: 0003:1532:007E.0015
2023-04-04 15:05:43 | razer.device0            | INFO           | Initialising device.0 RazerMouseDock
2023-04-04 15:26:44 | razer                          | INFO           | Initialising Daemon (v3.2.0). Pid: 2641
2023-04-04 15:26:44 | razer.screensaver    | INFO           | Initialising DBus Screensaver Monitor
2023-04-04 15:26:44 | razer                          | INFO           | Found device.0: 0003:1532:007E.0008
2023-04-04 15:26:44 | razer.device0            | INFO           | Initialising device.0 RazerMouseDock
2023-04-04 15:26:45 | razer                          | INFO           | Serving DBus
2023-04-04 15:26:47 | razer                          | WARNING  | Removing 0003:1532:007E.0008
2023-04-04 15:27:06 | razer                          | INFO           | Found valid device.0: 0003:1532:007E.0015
2023-04-04 15:27:06 | razer.device0            | INFO           | Initialising device.0 RazerMouseDock

`ls -al /sys/bus/hid/drivers/razer*/*1532*/` (nothing for 1532:0090. Does this mean the driver is missing? I've updated and rebooted)

'/sys/bus/hid/drivers/razeraccessory/0003:1532:007E.0015/':
total 0
drwxr-xr-x 5 root plugdev    0 Apr  4 15:27 .
drwxr-xr-x 5 root root       0 Apr  4 15:27 ..
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 country
-rw-rw---- 1 root plugdev 4096 Apr  4 15:27 device_mode
-r--r----- 1 root plugdev 4096 Apr  4 15:27 device_serial
-r--r----- 1 root plugdev 4096 Apr  4 15:27 device_type
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 driver -> ../../../../../../../../../../bus/hid/drivers/razeraccessory
-r--r----- 1 root plugdev 4096 Apr  4 15:27 firmware_version
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 hidraw
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 input
-rw-rw---- 1 root plugdev 4096 Apr  4 15:27 matrix_brightness
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_effect_none
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Apr  4 15:27 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 modalias
drwxr-xr-x 2 root plugdev    0 Apr  4 15:27 power
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 subsystem -> ../../../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Apr  4 15:27 test
-rw-r--r-- 1 root plugdev 4096 Apr  4 15:27 uevent
-r--r----- 1 root plugdev 4096 Apr  4 15:27 version

'/sys/bus/hid/drivers/razeraccessory/0003:1532:007E.0016/':
total 0
drwxr-xr-x 5 root plugdev    0 Apr  4 15:27 .
drwxr-xr-x 5 root root       0 Apr  4 15:27 ..
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 country
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 driver -> ../../../../../../../../../../bus/hid/drivers/razeraccessory
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 hidraw
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 input
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 modalias
drwxr-xr-x 2 root plugdev    0 Apr  4 15:27 power
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 subsystem -> ../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Apr  4 15:27 uevent

'/sys/bus/hid/drivers/razeraccessory/0003:1532:007E.0017/':
total 0
drwxr-xr-x 5 root plugdev    0 Apr  4 15:27 .
drwxr-xr-x 5 root root       0 Apr  4 15:27 ..
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 country
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 driver -> ../../../../../../../../../../bus/hid/drivers/razeraccessory
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 hidraw
drwxr-xr-x 3 root plugdev    0 Apr  4 15:27 input
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 modalias
drwxr-xr-x 2 root plugdev    0 Apr  4 15:27 power
-r--r--r-- 1 root plugdev 4096 Apr  4 15:27 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Apr  4 15:27 subsystem -> ../../../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Apr  4 15:27 uevent

`sudo apt list | ack razer`

openrazer-daemon/jammy,jammy,now 3.2.0+dfsg-3 all [installed]
openrazer-doc/jammy,jammy 3.2.0+dfsg-3 all
openrazer-driver-dkms/jammy,jammy,now 3.2.0+dfsg-3 all [installed]
openrazer-meta/jammy,jammy,now 3.2.0+dfsg-3 all [installed]
python3-openrazer/jammy,jammy,now 3.2.0+dfsg-3 all [installed]
qrazercfg-applet/jammy,jammy 0.42+ds-1build1 all
qrazercfg/jammy,jammy 0.42+ds-1build1 all
razercfg/jammy,now 0.42+ds-1build1 amd64 [residual-config]

nightsky30, 2023-04-04:
Openrazer v3.2.0 doesn't have support for the Naga Pro. It was added later. Try updating to the latest version.

c0d3m0nky, 2023-04-05:
Didn't think to check if the official Ubuntu repo wasn't actually the latest version, woops. Thanks for pointing that out

closed 2023-04-05 by c0d3m0nky