openrazer/openrazer #2044: Support for BlackWidow V4 Pro (Wired)

JimBlaney, 2023-04-08:
### Device Info

* Name: BlackWidow V4 Pro (Wired)
* Product Number: RZ03-04681900-R3U1
* Model Number: RZ03-0468

### Device Information
```
Bus 001 Device 006: ID 1532:028d Razer USA, Ltd Razer BlackWidow V4 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028d 
  bcdDevice            1.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
[... 176 lines cut ...]
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

001:006:000:DESCRIPTOR         1680922604.287714
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
More than happy to provide w/ coord

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
I don't have a WIndows machine to get this information.

Jacajack, 2023-04-15:
I've spent some time playing around with Wireshark and modifying the driver (https://github.com/jacajack/openrazer/tree/blackwidow-v4) in attempt to get the BlackWidow V4 Pro working. At the moment, I'm having problems with  `razer_set_device_mode()`. The keyboard appears to respond with random garbage instead of sending back the same payload. 

```
[ 3541.581304] razerkbd 0003:1532:028D.0018: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-8/input0
[ 3541.854550] razerkbd 0003:1532:028D.0019: input,hidraw2: USB HID v1.00 Keyboard [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-8/input1
[ 3541.907852] razer_set_device_mode()
[ 3541.907853] razer_chroma_standard_set_device_mode(0, 0)
[ 3541.907855] razer_send_payload()
[ 3541.907856] razer_get_report()
[ 3541.908787] razer driver: Device data transfer failed.
[ 3541.908866] razerkbd: Response doesn't match request. status: 49 transaction_id.id: 54 remaining_packets: 2000 protocol_type: a9 data_size: 4f, command_class: 00, command_id.id: 20 Params: 1d510020010000000000000000000000 .
[ 3541.908871] razerkbd: This was the request. status: 00 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 00, command_id.id: 04 Params: 00000000000000000000000000000000 .
[ 3541.909128] razerkbd 0003:1532:028D.001A: input,hidraw3: USB HID v1.00 Mouse [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-8/input2
[ 3541.909247] razerkbd 0003:1532:028D.001B: No inputs registered, leaving
[ 3541.909287] razerkbd 0003:1532:028D.001B: hidraw4: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-8/input3
[ 3542.031228] razerkbd 0003:1532:028D.001C: input,hiddev97,hidraw5: USB HID v1.00 Device [Razer Razer BlackWidow V4 Pro] on usb-0000:00:14.0-8/input4
```

The "set mode" command format appears to be correct, however. When Synapse is shutting down, it sends an identical command to the keyboard and the response is valid. Either I missed something in the driver, or setting mode is allowed only in certain context. 

This is my Wireshark dump: [blackwidow_v4_synapse_init_and_shutdown.tar.gz](https://github.com/openrazer/openrazer/files/11239937/blackwidow_v4_synapse_init_and_shutdown.tar.gz). You can filter by `(usb.bus_id == 1 && usb.device_address == 5) && razer`. It captures the moment I plugged the keyboard into the Windows VM and then shut down Synapse a few seconds later. At the very end you can see the "set mode" command.

z3ntu, 2023-04-22:
Try a different codepath in the `razer_get_report` method, that might fix it.

Jacajack, 2023-04-22:
Thanks for the tip! I've added Blackwidow V4 to the same switch branch as `USB_DEVICE_ID_RAZER_BLACKWIDOW_V3` and it appears to be working now. I can already set device mode, matrix brightness and read serial number. I'll see if I can get the rest of the features working.

Jacajack, 2023-04-22:
All matrix effects appear to be working "out of the box". I've only noticed that the wave effect direction is reversed. Custom effects work as well (including the LED stripe around the keyboard).

Blackwidow V4 has an additional "wheel" effect which doesn't seem to be supported by the driver. I also couldn't get the macro LED to turn on with the default commands. I'll see what I can do about that.

z3ntu, 2023-04-23:
For wave direction, we have `WAVE_DIRS` which setting to a different value might fix that. If that doesn't help you can try something like https://github.com/openrazer/openrazer/blob/master/driver/razeraccessory_driver.c#L672

clveranis, 2023-06-08:
Bumping for attention..

I know that people are still working on this [PR - 2054](https://github.com/openrazer/openrazer/pull/2054) but it seems like at least the RGB portion for the BWv4 is figured out. Would we be able to get that functionality merged? that would be awesome. 

Thanks!

Jacajack, 2023-06-11:
That's right - RGB control in the driver part seems to be working well, but I still haven't made necessary changes one the Python daemon side (the "wheel" effect needs to be added). I can focus on that instead of trying to get the macro LED working.

Personally, I'm fine with providing support for BlackWidow V4 in a few smaller PRs. I'm not sure what the maintainers think about that, though.

renatojf, 2023-10-02:
Can we please get the PR merged? :pleading_face: