openrazer/openrazer #2045: Steam Deck: 1532:009C and 1532:0271 not working via Bluetooth

vazome, 2023-04-08:
### Distribution

Linux steamdeck 5.13.0-valve36-1-neptune #1 SMP PREEMPT Mon, 19 Dec 2022 23:39:41 +0000 x86_64 GNU/Linux

### Support

- [yes] Have you tried rebooting?
- [yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Installation [script](https://daniele.tech/2022/12/how-to-get-openrazer-working-easily-on-steam-deck/) used:
<details> 
 ``` shell
#!/bin/bash
sudo steamos-readonly disable
sudo sed -i "s/SigLevel    = Required DatabaseOptional/SigLevel = Never/g" /etc/pacman.conf
sed -i "s/LocalFileSigLevel = Optional/#LocalFileSigLevel = Optional/g" /etc/pacman.conf
sudo pacman-key --refresh-keys
sudo pacman-key --init
sudo pacman -Sy archlinux-keyring
sudo pacman-key --populate archlinux
sudo pacman -Syu base-devel mlocate cmake gcc holo-rel/linux-headers jupiter-rel/linux-neptune-headers holo-rel/linux-lts-headers $(pacman -Qk 2>/dev/null | grep -ve ' 0 missing' | grep -ie ^libc -e glibc -e gcc -e clang -e headers -e udev -e systemd | awk -F ':' '{print $1}') --overwrite '*'
cd /tmp/
git clone https://aur.archlinux.org/openrazer.git
cd openrazer
makepkg -sri
systemctl --user enable --now openrazer-daemon.service
sudo gpasswd -a deck plugdev
```
</details>

Once I installed the drivers on to my Steam Deck I have encountered an issue, the all of my Razer devices are not identified when connected via Bluetooth:
- Razer BlackWidow V3 Mini Hyperspeed (BT, 2.4) – Mentioned in supported devices as (Wireless)
- Razer DeathAdder V2 X HyperSpeed (BT, 2.4, USB) – Mentioned in supported devices

NOTE: Everything works well over 2.4 receiver connection, meaning devices are showing in RazerGenie.

``` shell
 #2.4
(1)(deck@steamdeck ~)$ lsusb | grep '1532:'
Bus 001 Device 004: ID 1532:009c Razer USA, Ltd Razer DeathAdder V2 X HyperSpeed

#Bluetooth
Device FD:C4:56:F9:59:41 (random)
        Name: DA V2 X
        Connected: yes
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
```

``` shell
#2.4
(127)(deck@steamdeck ~)$ lsusb | grep '1532:' 
Bus 001 Device 005: ID 1532:0271 Razer USA, Ltd Razer BlackWidow V3 Mini

#Bluetooth
Device E0:3C:7F:4A:5D:AF (random)
        Name: BlackWidow V3 Mini
        Connected: yes
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)

z3ntu, 2023-04-22:
Potentially the udev rules didn't trigger so the driver wasn't rebount.

Can you check `/sys/bus/hid/drivers/razer*/` ?

vazome, 2023-04-25:
Just for reminder, the issue is only related to Bluetooth connection, because it works okay via 2.4 (USB Receiver):
```
/sys/bus/hid/drivers/razerkbd/:
total 0
drwxr-xr-x 2 root root    0 Apr 26 00:13 .
drwxr-xr-x 7 root root    0 Apr 26 00:13 ..
--w------- 1 root root 4.0K Apr 26 00:20 bind
lrwxrwxrwx 1 root root    0 Apr 26 00:21 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Apr 26 00:20 new_id
--w------- 1 root root 4.0K Apr 26 00:13 uevent
--w------- 1 root root 4.0K Apr 26 00:20 unbind

/sys/bus/hid/drivers/razermouse/:
total 0
drwxr-xr-x 2 root root    0 Apr 26 00:13 .
drwxr-xr-x 7 root root    0 Apr 26 00:13 ..
--w------- 1 root root 4.0K Apr 26 00:20 bind
lrwxrwxrwx 1 root root    0 Apr 26 00:21 module -> ../../../../module/razermouse
--w------- 1 root root 4.0K Apr 26 00:20 new_id
--w------- 1 root root 4.0K Apr 26 00:13 uevent
--w------- 1 root root 4.0K Apr 26 00:20 unbind
```

johnjohnlin, 2023-06-22:
I have DA V2 X as well, and I want to ask an extra problem. I am using archlinux but I cannot connect to the device using bluetoothctl, and 2.4G works fine. I am not sure whether this issue is related to the openrazer driver or not?

My system information:

```text
Kernel is 6.3.8-arch1-1.
BT is Intel AC9260.
```

Bluetooth output
```text
[bluetooth]# pair CD:6E:EA:A7:B8:2B
Attempting to pair with CD:6E:EA:A7:B8:2B
[CHG] Device CD:6E:EA:A7:B8:2B Connected: yes
Failed to pair: org.bluez.Error.ConnectionAttemptFailed
[CHG] Device CD:6E:EA:A7:B8:2B Connected: no
[bluetooth]# info CD:6E:EA:A7:B8:2B
Device CD:6E:EA:A7:B8:2B (random)
        Name: DA V2 X
        Alias: DA V2 X
        Appearance: 0x03c2
        Icon: input-mouse
        Paired: no
        Bonded: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
```

closed 2023-11-20 by github-actions[bot]