openrazer/openrazer #2046: Support for Kraken Kitty Edition V2 Pro

printfn, 2023-04-10:
### Device Info

* Name: Razer Kraken Kitty Edition V2 Pro
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Bus 001 Device 003: ID 1532:0554 Razer USA, Ltd Razer Kraken Kitty V2 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0554 
  bcdDevice            0.01
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken Kitty V2 Pro
  iSerial                 3 000000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0141
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 USB Audio Device
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0071
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
[... 313 lines cut ...]
```

<details>

```
001:003:003:DESCRIPTOR         1681101041.792005
 05 0C 09 01 A1 01 85 01 15 00 26 FF 00 09 00 75
 08 95 3D 91 02 09 00 81 02 85 04 09 00 75 08 95
 1A 91 02 85 05 09 00 75 08 95 16 81 02 85 08 09
 00 75 08 95 01 81 02 85 40 09 00 75 08 95 0E 91
 02 85 41 09 00 75 08 95 0E 81 02 85 62 09 00 75
 08 95 0F 91 02 85 63 09 00 75 08 95 0F 81 02 85
 52 15 00 25 01 09 E9 09 EA 75 01 95 02 81 06 09
 00 95 06 81 01 C0 05 0B 09 05 A1 01 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

ethanbonin, 2023-07-02:
Hey! I'm new to this community and hardware support. I don't know how helpful I will be, but what can I do to help with the Kraken Pro V3?

z3ntu, 2023-07-16:
@ethanbonin Apart from actually adding support for it (see [DEVELOPMENT.md](https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md)), capturing usb packet capture would be useful for reference. https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

printfn, 2023-07-22:
I've updated the post to add some Synapse screenshots of the various options, and I'll see if I can set up packet capture soon.

paladin-devops, 2025-01-17:
@printfn any update on a pcap for this one?

Christoph8300, 2025-09-02:
Hi i just found this issue and i also want my razer kraken kitty v2 pro in openrazer. I just made a pcap file with wireshark. I hope it contains the right informations. 

[Razer_Kraken_Kitty_V2_Pro.zip](https://github.com/user-attachments/files/22093634/Razer_Kraken_Kitty_V2_Pro.zip)