openrazer/openrazer #2048: Settings won't apply correctly after reboot

Kyr4l, 2023-04-15:
### Distribution
OpenSUSE Tumbleweed

### Support

- [X ] Have you tried rebooting?
- [X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Each time i reboot my computer, the last config doesn't apply, intead a new row appears in `.config/openrazer/persistence.conf` : 
This is my current config
```
(0)-$ cat .config/openrazer/persistence.conf 
[MYMOUSESERIAL]
dpi_x = 1300
dpi_y = 1300
poll_rate = 1000
```

And this is what happens once i reboot :
```
(0)-$ cat .config/openrazer/persistence.conf 
[MYMOUSESERIAL]
dpi_x = 1300
dpi_y = 1300
poll_rate = 1000

[UNKWN000000000489]
dpi_x = 1800
dpi_y = 1800
poll_rate = 500
```

Each reboot adds an [UNKNWN] string to the config

lah7, 2023-04-15:
I believe that some Razer devices don't always respond with their serial number when requested, so the driver fallbacks to `UNKNWN`. Which mouse is this?

A workaround for now could be to add a command (or write a script) that runs at log on to apply the desired settings. Here's an example with the Python library as a one liner command:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager(); a.devices[0].dpi = (1300,1300); a.devices[0].poll_rate = 1000"

Kyr4l, 2023-04-17:
the mouse i use is a Deathadder V2 X Hyperspeed

also this issue feels random, yesterday i booted up my computer and the settings were correctly applied, today i booted it up again and the issue came back

closed 2023-09-15 by github-actions[bot]