openrazer/openrazer #2049: mouse doesn't work on boot

Agreeable-f, 2023-04-15:
### Distribution

OS: Arch linux 
Kernel: 6.2.10-zen1-1-zen
DE: KDE Plasma 5.27.4
WM: KWin (X11)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Upon booting and logging into my pc my mouse doesn't work, no mouse movement, and isn't recognized by tools like polychromatic or razergenie. When I unplug and re-plug my mouse it works again.

side note but the sniper button doesn't do anything and isn't mappable by inputmapper. i'd like to be able to map it

lah7, 2023-04-15:
Does this problem happen even when OpenRazer is uninstalled?

If so, then it might be a BIOS quirk. Does it happen when plugged in to other USB ports? You may want to check your BIOS for any USB settings that can help defer initialization of USB devices until the OS is booted, like turning off "Legacy USB Support" if it's enabled.

Not sure what mouse has a "sniper button", but if something like the "Hypershift" feature, it's not currently supported in OpenRazer yet (#2031)

Agreeable-f, 2023-04-16:
I ran `sudo pacman -R openrazer-meta` then restarted.
same issue. the sniper button is exclusive to razer's basilisk series that i know of for razer mice. it's just another mouse button that *should* be programmable but i think functions as a hypershift or dpi shift button by default.

will update again if i find something in my bios that may be causing this.

Agreeable-f, 2023-04-16:
I've tried disabling "Legacy USB" in bios. both with and without openrazer installed same issue occurs. also on different usb ports on my mobo still getting this issue.

lah7, 2023-04-16:
Unless the _orphaned_ packages were removed afterwards, uninstalling `openrazer-meta` alone probably didn't _actually_ remove the driver/daemon. Try:

    sudo pacman -R openrazer-meta openrazer-driver-dkms openrazer-daemon python-openrazer

If your BIOS is the modern UEFI interface kind, does the mouse work there too? Try replugging the mouse while the computer is turned off too, so the mouse in theory should be in a "standard" state - just to rule out if it is OpenRazer's fault or some kind of hardware/BIOS quirk.

Agreeable-f, 2023-04-16:
I ran `sudo pacman -R openrazer-meta openrazer-driver-dkms openrazer-daemon python-openrazer` to uninstall openrazer.

mouse still doesn't move on boot however i did find some odd behaviour from it. 

the mouse doesn't work in the BIOS by default however will work if i re-plug my mouse when my pc is off or in BIOS. it doesn't work when i actually boot into linux however. both with and without Legacy USB.

lah7, 2023-04-17:
Unfortunately, since it doesn't work in the BIOS where there's no driver present, it's sounds more like a hardware problem then something OpenRazer can fix.

We haven't checked to see if Linux physically detects its presence:

    lsusb | grep Razer

In some cases, Linux might also complain about USB devices, which might offer a clue:

    sudo dmesg | grep usb

Out of interest, what is your BIOS?

    sudo dmidecode | head

It wouldn't be the first time I heard that some Razer hardware (usually a USB hub) acting erratic under some BIOSes.  Some other things to try:

* In the BIOS, if there are any power saving settings, see if you can get the USB devices to be fully powered "off" when it's shut down. It seems power cycling the mouse makes it work again?
* What about restarting? Is this only an issue when the system is _shut down_?
* See if there's a BIOS update.
* Use a Windows PC and Razer Synapse to see if there is a firmware update for your Razer Basilisk.

Agreeable-f, 2023-04-17:
### I ran these commands post re-plugging. Will update with outputs pre re-plugging

#### Results from `sudo lsusb | grep Razer`

```
Bus 002 Device 009: ID 1532:0226 Razer USA, Ltd Huntsman Elite
Bus 002 Device 015: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3
```

#### Results from `sudo dmesg | grep usb`

Uploaded to gist, it's a long output.
https://gist.github.com/Agreeable-f/94e15e26bebeb3c0db81048c069f6a62

#### Results from `sudo dmidecode | head`

```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.
Table at 0x2FD3D000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: American Megatrends Inc.
	Version: F8
	Release Date: 10/15/2019
```
i'm sure this is an issue with the controller and/or firmware on the mouse itself. ~~I wonder if it gets stuck at trying to install synapse or whatever it does to set itself up and panics.~~

Agreeable-f, 2023-04-17:
### I ran these commands pre re-plugging

#### Results from `sudo lsusb | grep Razer`

> Bus 002 Device 009: ID 1532:0226 Razer USA, Ltd Huntsman Elite

The mouse (basilisk v3) isn't detected on boot

#### Results from `sudo dmesg | grep usb`

Uploaded to gist. it's a long output.
https://gist.github.com/Agreeable-f/700b7cdd1a1953b9f6cce5c2987f35c5

#### Results from `sudo dmidecode | head`

```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.1.1 present.
Table at 0x2FD3D000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: American Megatrends Inc.
	Version: F8
	Release Date: 10/15/
```

Included this one just in case and for uniformity.

I'll update with results from BIOS power settings, though i doubt it'll affect anything since it *does* work in bios after being re-plugged however not on OS boot.

This also happens on restart. 

I'm sure there's a bios update available. ~~i've never updated my mobo BIOS before~~

i don't have a spare pc running windows with synapse installed, i'll have to get back on that one.

lah7, 2023-04-17:
Ah, the Razer Basilisk **V3**... looks like there were some issues about this before:

- #1844
- #1767

Apparently the hardware effect _state_ is the issue. One reported a [factory reset](https://github.com/openrazer/openrazer/issues/1767#issuecomment-1461519512) under Synapse (on Windows) did the trick. Another found that [leaving it on Spectrum](https://github.com/openrazer/openrazer/issues/1767#issuecomment-1107477560) works. Apparently, there's [a firmware update](https://github.com/openrazer/openrazer/issues/1767#issuecomment-1280919628) -- be good to know what yours is running now.

Sorry to your mobo, it's not the BIOS' fault -- just buggy Razer firmware :)

Technically, you could use a [Windows VM](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/) (or [ISO](https://www.microsoft.com/en-gb/software-download/windows11/)) and passthrough the USB mouse, this is **not recommended** for firmware updates (and Synapse might do it automatically) as it risks bricking the device from a lost connection. Something bare metal like a Windows To Go bootable USB drive would be better, but still going to need to borrow a Windows installation anyhow.

Agreeable-f, 2023-04-17:
Those issues seem to be *after* having installed openrazer. Not blaming them but it seems it causes some odd state on the mouse itself.  

Something odd to note is that i have OpenRGB-git installed which can't find the mouse at all even though it's listed as compatible and it's enabled under supported devices. My keyboard (huntsman elite) now doesn't save it's rgb profile either, though i can still edit the rgb from OpenRGB.  I did try and change the rgb from razergenie to static red (to match my setup ofc) and after a reboot it has no rgb at all. though i did also uninstall razergenie since then. Possibly the RGB state having a panic? i'm not completely sure i'm just throwing out ideas. Can't even reset the effect state on boot since it isn't recognized. 

Would it be possible to patch the firmware itself? It would eliminate the issue of buttons not being easily bound and odd quirks like this but it seems like more work than it's worth and i'm not knowledgeable about this stuff so i'd like to hear your thoughts about it/learn about it more.

lah7, 2023-04-18:
It's plausible that OpenRazer triggered this buggy state - it is using _unofficial_ hardware features of the mouse that were analysed from older Razer generations. Also possible that the hardware revision is different to those who added support originally. Officially, Razer Synapse on Windows handles _practically everything_ in software, especially true for **Synapse 3** era of hardware, like Basilisk V3.

With that in mind, I wonder if setting a "custom" (software) effect on the mouse helps this problem, as opposed to leaving a hardware effect set (Static, Wave, Breath, etc), which are technically _unsupported_ by Razer. If you don't have a GUI effect editor handy that can paint a colour on all the keys individually (like in Polychromatic or RazerGenie), you can run this from command line:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager(); b = devices[0]; print(b.name); b.fx.advanced.matrix[0,0] = (255,0,0); b.fx.advanced.draw();" 

> (Change `devices[0]` to `devices[1]` if your keyboard is first)

Custom/software effects don't persist between power cycles -- they _could_, but the driver doesn't do that (`VARSTORE`) to avoid wearing out the flash memory. I guess this might not be supported on all firmware anyway. The workaround could be to have a Python script running at login to "draw" on all the individual keys in the matrix, effectively emulating the static effect but through the "custom" (software) one. In theory, it's the hardware effect(s) that triggers buggy behaviour.

Unfortunately, this is all I can probably help with. I'm not familiar with fixing the hardware/firmware/driver side. I focus on developing [a frontend](https://github.com/polychromatic/polychromatic) (of which software driven effects is on the radar) so this issue is good to know.

Agreeable-f, 2023-04-19:
I created an effect in the effects tab of polychromatic. It changes all the lights to red (127) for all led's and that's about it. The mouse still isn't recognized after a restart.

#### Results from `sudo dmesg | grep usb` just in case

Uploaded to gist. it's a long output.
https://gist.github.com/Agreeable-f/afbe0c7e608d4f0af9a34b6aa73d73bf

nogaff, 2023-06-12:
I have the same issue with my Basilisk V3, where it is not detected after a reboot until it gets replugged, but I noticed something else which nobody seems to have mentioned yet; if you press the button on the underside of the mouse to switch to a different onboard profile prior to a shutdown/reboot, the mouse is then detected correctly during the next boot and you can subsequently switch back to the default profile with your preferred RGB settings, (until you need to shut down again).

It's no less irritating than replugging, but I just thought it might provide another clue if anyone's still investigating the issue.

Of course, this requires you to have actually saved an onboard profile with Razer Synapse before using OpenRazer.

favonia, 2023-11-04:
I'd like to add a data point: I already upgraded the firmware to the latest version, and my situation is still exactly the same as @quietvoid's https://github.com/openrazer/openrazer/issues/1767#issuecomment-1107477560. That is, setting effects to None triggered the bad state but setting it back to Spectrum with 0% brightness worked. Visually I can be satisfied with 0% brightness.

@lah7 So, the new firmware still did not (completely) fix it. 😟 On the other hand, I wonder if the frontend can give a warning about various suspicious configurations that have been reported to break the mouse? Maybe it can even suggest the 0% brightness hack. In my case, setting _any_ effect (mouse, scroll wheel, logo) to None could potentially trigger the firmware bug.

nogaff, 2023-11-04:
> I'd like to add a data point: I already upgraded the firmware to the latest version, and my situation is still exactly the same as @quietvoid's [#1767 (comment)](https://github.com/openrazer/openrazer/issues/1767#issuecomment-1107477560). That is, setting effects to None triggered the bad state but setting it back to Spectrum with 0% brightness worked. Visually I can be satisfied with 0% brightness.

Can confirm. I just gave this a try and it worked fine for me after a reboot, having set the effect to Static along with 0% brightness and/or black for the colour, so it certainly seems like the root cause of the issue is the effect being set to None.

> I wonder if we can give a warning about various suspicious configurations that have been reported to be able to break the mouse? Maybe it can even suggest the 0% brightness hack. In my case, setting any effect (mouse, scroll wheel, logo) to None could potentially trigger the firmware bug.

I feel like the onus is mostly on the various frontend apps to provide that kind of information. I'd say the OpenRazer driver should probably disallow setting the effect to None on devices where it's known to cause problems, but the frontend apps should inform the user as to why that's the case, and explain the workaround.

lah7, 2023-11-04:
Thanks for the feedback. I think it would be wise to just remove the "None" effect for this mouse from OpenRazer, so _nothing_ can accidentally trigger the buggy firmware. Opened a PR:

* #2156

From what I remember, "None" was an effect from the Synapse 2 era of devices - so I guess it's really just an unsupported leftover in the firmware that the official Synapse drivers would never use.

favonia, 2023-11-04:
@lah7 Thank you! In case it helps, I noticed that a suspend (sleep) is enough to demonstrate the bug in my case. A reboot is not needed (in a bad way).

closed 2023-11-25 by z3ntu

AutomatiqueTango, 2026-01-11:
That bug is happening to me too.
Unplug/replug it is the only fix that I found.
My mouse (Basilisk V3) is updated to the latest firmware.
I only have RazerGenie installed.

commit 606ed017 (2023-11-25): Remove "None" effect for Basilisk V3

Setting the "None" effect causes this mouse to enter a buggy state.
If the computer was rebooted while the "None" is set, the mouse won't be
detected again until it is physically replugged.

Some users found that a firmware update solved it, while others reported
a factory reset (on Synapse) works. Ultimately, the firmware is stable
when none of the "None" effects are set.

Better to prevent this situation from happening by removing the effect.

Fixes #2049
Addresses #1767, #1844
PR #2156 (2023-11-25): Remove "None" effect for Basilisk V3
Setting the "None" effect causes this mouse to enter a buggy state. If the computer was rebooted while the "None" is set, the mouse won't be detected again until it is physically replugged.

Some users found that a firmware update solved it, while others reported a factory reset (on Synapse) works. Ultimately, the firmware is stable when none of the "None" effects are set.

Better to prevent this situation from happening by removing the effect.

Fixes #2049
Addresses #1767, #1844