openrazer/openrazer #2053: Support for [Razer Ornata V3]

Tankeran, 2023-04-24:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
 Bus 001 Device 004: ID 1532:02a1 Razer USA, Ltd Razer Ornata V3
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a1 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 244 lines cut ...]
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:003:000:DESCRIPTOR         1682354721.976687
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-04-24:
This is like the third version of "Ornata V3" with a different USB PID again. Is the device really just called "Ornata V3" or with some other suffix?

Can you please also find the Product Number & Model Number on the device and send them?

Tankeran, 2023-04-26:
Hello,

Attached you may find product details.

[image]

z3ntu, 2023-04-26:
Thanks, I've added the info to the initial comment. Since the model number (RZ03-0446) is the same as #1925 but product number is higher, maybe it's a newer revision of the device? But in any case from openrazer side it's a different USB PID so it needs separate device support.

z3ntu, 2023-04-27:
USB packet captures from Razer Synapse on Windows would be helpful to have https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

But otherwise not really I think.

Tankeran, 2023-04-28:
Hello,
Unfortunately I dont have Windows computer. Only have Steam deck :S

Evonder, 2023-04-30:
Oh man I was so excited when this was merged and released. I updated on Debian this afternoon but noticed it was still unrecognized, then I noticed I have yet another version of the Ornata V3 X (I know this thread is for the non X ver.). My version is:
```Bus 004 Device 004: ID 1532:02a2 Razer USA, Ltd Razer Ornata V3 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a2 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3 X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

z3ntu, 2023-05-01:
@Evonder Please open a new issue for that device, thanks.

jackwilliams79, 2023-05-02:
@Evonder and @z3ntu I've created a new issue for the same device (Razer Ornata V3 X) and same product id (02a2). Please have a look at https://github.com/openrazer/openrazer/issues/2059

And sorry for the offtopic.

Tankeran, 2023-05-21:
any progress on this topic?

greasydeal, 2023-07-16:
[OrnataV3(0x02A1).tar.gz](https://github.com/openrazer/openrazer/files/12064662/OrnataV3.0x02A1.tar.gz)

Let me know if these will work.

z3ntu, 2023-07-17:
Seems fine! The device uses transaction id 0x1f, and uses `razer_chroma_extended_matrix_effect_*` methods and looks like `MATRIX_DIMS = [1, 10]` I believe.

Other than that there's not more info in the pcaps. See https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how to add support for this device.

greasydeal, 2023-07-17:
> Seems fine! The device uses transaction id 0x1f, and uses `razer_chroma_extended_matrix_effect_*` methods and looks like `MATRIX_DIMS = [1, 10]` I believe.
>
> Other than that there's not more info in [...]

I actually forked the project yesterday morning on have my V3 working in my test VM right now. I still need to work out reactive and starlight modes. Should we a PR later this week.

xandervill, 2023-08-02:
Hello. 
In applications Polychromatic and RazerGenie Ornata V3 it is not defined. Will there be keyboard Ornata V3 support for Linux Ubuntu 22.04? If not, is it possible to use drivers from Ornata V2 and how to do it?
[image]

z3ntu, 2023-08-13:
Someone opened (and closed again) a pull request for this device: #2096 Not sure what the state there is.

Tankeran, 2023-08-15:
hello friends i just saw there is already shared document by @greasydeal unfortunately I dont know how to try that driver. Could you guide me through?

skyline-ch, 2023-08-25:
Hello, I have just acquired the Ornata V3 keyboard, I am under Ubuntu 20.04.
The keyboard works well, but it is not recognized by Polychromatic, from what I see, it is due to the version which is not supported by Open Razer (too bad I had checked before buying that I find it in the list)

I can provide you with information from a virtual machine if necessary, or from my Linux, do tests with a little help for complex manipulations.

given on the keyboard:
model no. RZ03-0446
P/NO. RZ03-04460900-R3Q1

PR #2182 (2024-04-09): Add Support for Razer Ornata V3 (02A1)
This adds support for the Razer Ornata V3, finalizing the bulk of the work from #1940.

With adaptions from the PR, I noticed that the static and spectrum modes turn off on a key press only when the reactive effect was set at least once beforehand. Upon replugging the keyboard while persistently configured to display the static effect, the issue does not occur so long as the effect is not switched to reactive at any point when it is plugged in. This behavior is even consistent across replugs and seems to have been missed in the linked PR.

Since deeming t [... cut]
commit 6c80dfd6 (2024-04-14): Add support for Razer Ornata V3

Fixes #2053
closed 2024-04-09 by z3ntu